PX4/PX4-Autopilot #7222: Access denied to connect to PX4flow in Comm Links. What is the output?

grantjerry, 2017-05-12:
Hello everyone, 

I want to test the PX4flow camera I just bought. I basically connected it with Micro USB cable to the PC, on which QGroundControl v3.1.3 was installed. I followed the official guideline to upgrade the firmware (stable version via QGC) and windows driver.   

Now, in the "Vehicle Setup" I can see the video of PX4Flow that is capturing. But I don't see anything like "OPTICAL_FLOW.flow_x" or "OPTICAL_FLOW_RAD.integraded_x" and so on in the "Analyze" plot window in QGC. If I try to create a Comm Link with Serial and COM3 (recognized on my PC), and click connect, it will show "QGCMessageBox (unit testing) "Link Error" "Error on link \\\\.\\COM3. Error connecting: Could not create port. Access is denied." How do I get to check the output of the PX4Flow?

Besides translations, does it output rotation and distance? What are their units?

grantjerry, 2017-05-16:
Hello Could anybody help?

jimchenhub, 2017-06-26:
Hello, Have you solved this problem?

LorenzMeier, 2017-08-20:
Closing as stale - please open a new issue if this persists with a recent version of QGC.

closed 2017-08-20 by LorenzMeier

drtrigon, 2021-01-21:
I am currently working on a research project where we want to have several kinds of sensors on a flying platform. The project team has moved to another university, and we struggle currently to get the full setup up and running, namely the PX4Flow camera. We do have this issue:

"Error on link PX4 Flow on COM?? (AutoConnect). Error connecting: Could not create port. Das System kann die angegebene Datei nicht finden."

I do also see that actually whenever I connect it 2 COM ports do get created. I guess this is already a sign that something does not work as expected, right?