PX4/PX4-Autopilot #7223: Mavlink threadding analysis

bkueng, 2017-05-12:
I tried to do a thorough analysis of potential concurrency issues in the mavlink module, since we have experienced random mission upload failures, log download issue, and other things from time to time.

Current implementation:
- 1 thread per sender and receiver
- Each mavlink instance starts a sender & a receiver
- There is already a lock in place, namely in `Mavlink::begin_send()`, which is taken into account for the analysis below

## Problems
1. Within the mavlink protocol parser: access to the same data from `mavlink_parse_char` (receiver thread) and `mavlink_msg_*send_struct` methods (main & receiver thread): RW access from both to the status struct (returned via `mavlink_get_channel_status()`):
   - `signing->timestamp`: a 64bit value
   - `mavlink_signing_streams_t` data structure (which is even shared between all mavlink instances)
   - `parse_error` counter
2. concurrent calls to `mavlink_msg_*send_struct` from the main & receiver thread (eg ping response, capabilities message, ...): RW access to the sequence counter `current_tx_seq`.
3. mavlink streams linked list: the receiver thread iterates this list (for reading the message interval), and it's possible that the main thread deletes an item from the list (when setting the rate to 0) at the same time leading to invalid memory access.
4. `get_free_tx_buf()` is called from the receiver and there is a race: the call could indicate enough free sending space, and in the meanwhile the main thread sends a message, filling the sending buffer, so that the receiver's message that should be sent will be dropped, although the receiver was under the impression there was enough space. This can happen for the FTP, log handler & param manager (and the other way around for the shell).
5. the mission manager, ftp, log handler & param manager are accessed from both threads: `handle_message()` and `send()` are both called from different threads and both modify various class members. This is the most severe problem and a lot can go wrong. It is a good candidate for the random errors we experienced.

## Solutions
1. The `parse_error` is just a status flag, and I think we can live with it being wrong with low probability (I don't see it used in PX4 anyway). The signing state is much more important and will need a proper solution as soon as we start to use signing. One solution is to add yet another lock (in a similar way as `MAVLINK_START_UART_SEND` is done, but we would also need to ensure proper ordering of the messages such that the timestamp is increasing).
2. A simple solution to that is to move the [MAVLINK_START_UART_SEND](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L348) call further up to line 304, since we take a lock within MAVLINK_START_UART_SEND. It would increase the critical section a bit, but the receiver thread does not send data often, so that it should be ok.
3. I think this is highly unlikely and I'd leave it as is for now.
4. The result here is the same as if data got dropped on the link and the protocols need to be able to handle it anyway, so it should not need any change.
5. A simple solution without the need for additional locking is to call everything from the same thread. And the easier thing is to call `send()` from the receiver thread regularly, so that the objects are never accessed from the main thread. This means the receiver would need to wake up more often, and the rate calculation/throttling would not take these into account anymore. It's probably not an issue, but if it is, we could still leave it as a stream and leave the `send()` empty, and use another method for the regular update from the receiver.

So only 2 and 5 call for immediate action, 1 when we want to use signing.

Probably related: #6984

tops4u, 2017-05-12:
Hi, could this be related? #6260 
I don't want to bring this to every Issue but I thought this might have an Impact on MavLink Stream Freezes?

bkueng, 2017-05-12:
I don't think so. I've never experienced such freezes, so I cannot say what exactly goes wrong in your case. I suggest you check if there is any data coming through at all during a freeze and look at `mavlink status`, in particular the `rate mult` value.

ecmnet, 2017-05-12:
Just to let you know, that in my case the stability of the communication between the companion which streams a lot of data to the controller (e.g. vision at 60Hz) and PX4 decreased at least since release 1.5. This affects mainly packages with large content, e.g. ULOG streaming and reading logs via MAVLink.

tops4u, 2017-05-12:
I observed the same that this happens more often since 1.5. I use a 3RD Radio on the same Serial Port as the OSD so this may cause Message congestion? I have never seen those Messages prior to 1.5 and now they happen very often and correspond to Logging Dropouts.

LorenzMeier, 2017-05-12:
@bkueng Can you move on to implement the basic, low hanging steps as proposed? I would do a re-architecture only as the 2nd step.

ecmnet, 2017-05-13:
Refer also to the effect of txBuffer-size increase: https://github.com/PX4/Firmware/pull/7226#issuecomment-301227157

closed 2019-02-13 by stale[bot]