PX4/PX4-Autopilot #7227: Unable to add a new airframe

AlexMalo, 2017-05-12:
I defined the files that you require in the directories init.d  and mixers, with the formats specified, under  ROMFS/px4fmu_common/ directory, using as template one of the files already in the directory. I see the new airframe in airframes.xml. However, when I run QGroundControl I do not see the airframe. How can I choose it? What I am missing?
Thanks
Alejandro Malo
[2107_wilga.txt](https://github.com/PX4/Firmware/files/997835/2107_wilga.txt)
[highfixedwing.main.mix.txt](https://github.com/PX4/Firmware/files/997837/highfixedwing.main.mix.txt)

LorenzMeier, 2017-05-12:
You need to flash the binary through QGroundControl so that it does pick up the new metadata.

closed 2017-05-12 by LorenzMeier

LorenzMeier, 2017-05-12:
I added additional instructions on this page: https://dev.px4.io/en/airframes/adding_a_new_frame.html