PX4/PX4-Autopilot #7228: Problems writing my first PX4 program

gesemeliab, 2017-05-12:
HI!
i'm new in PX4 development and i'm trying to write my first program, I've followed all the instructions from the PX4 dev guide but when i'm trying to build my program this comes out:  
[2/538] px4io-v2: nuttx-patches/00002-...ECTED-fix-shadow-wanings.patch applied
FAILED: cd /home/gesem/Firmware/build_px4fmu-v2_default/src/modules/px4iofirmware && /usr/bin/patch --verbose -d /home/gesem/Firmware/build_px4fmu-v2_default/px4io-v2/NuttX -s -p1 -N < /home/gesem/Firmware/nuttx-patches/00002-REJECTED-fix-shadow-wanings.patch && cmake -E touch /home/gesem/Firmware/build_px4fmu-v2_default/px4io-v2/NuttX/px4io-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch.stamp
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/signal.h NuttX/nuttx/include/signal.h
|index bd8ee9f..7ee242a 100644
|--- NuttX/nuttx/include/signal.h
|+++ NuttX/nuttx/include/signal.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/stdlib.h NuttX/nuttx/include/stdlib.h
|index aa259c9..f2b0640 100644
|--- NuttX/nuttx/include/stdlib.h
|+++ NuttX/nuttx/include/stdlib.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/sys/stat.h NuttX/nuttx/include/sys/stat.h
|index 3f2658b..41c8604 100644
|--- NuttX/nuttx/include/sys/stat.h
|+++ NuttX/nuttx/include/sys/stat.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/sys/statfs.h NuttX/nuttx/include/sys/statfs.h
|index 1d1786a..12f802b 100644
|--- NuttX/nuttx/include/sys/statfs.h
|+++ NuttX/nuttx/include/sys/statfs.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
[2/538] px4fmu-v2: nuttx-patches/00002...ECTED-fix-shadow-wanings.patch applied
FAILED: cd /home/gesem/Firmware/build_px4fmu-v2_default && /usr/bin/patch --verbose -d /home/gesem/Firmware/build_px4fmu-v2_default/px4fmu-v2/NuttX -s -p1 -N < /home/gesem/Firmware/nuttx-patches/00002-REJECTED-fix-shadow-wanings.patch && cmake -E touch /home/gesem/Firmware/build_px4fmu-v2_default/px4fmu-v2/NuttX/px4fmu-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch.stamp
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/signal.h NuttX/nuttx/include/signal.h
|index bd8ee9f..7ee242a 100644
|--- NuttX/nuttx/include/signal.h
|+++ NuttX/nuttx/include/signal.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/stdlib.h NuttX/nuttx/include/stdlib.h
|index aa259c9..f2b0640 100644
|--- NuttX/nuttx/include/stdlib.h
|+++ NuttX/nuttx/include/stdlib.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/sys/stat.h NuttX/nuttx/include/sys/stat.h
|index 3f2658b..41c8604 100644
|--- NuttX/nuttx/include/sys/stat.h
|+++ NuttX/nuttx/include/sys/stat.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git NuttX/nuttx/include/sys/statfs.h NuttX/nuttx/include/sys/statfs.h
|index 1d1786a..12f802b 100644
|--- NuttX/nuttx/include/sys/statfs.h
|+++ NuttX/nuttx/include/sys/statfs.h
--------------------------
File to patch: 
Skip this patch? [y] 
1 out of 1 hunk ignored
ninja: build stopped: subcommand failed.
Makefile:161: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

Does anyone know what is wrong and how to solve it ?

dagar, 2017-05-12:
Are you synced up with the latest version of master?
Try `make distclean` then `make`.

gesemeliab, 2017-05-12:
Thank you! it worked very well i think ! but can you explain me what is master ? and why do i needed to write make distclean and then make ? i want to understand a bit more of all of this haha
THANKS!

dagar, 2017-05-12:
I don't know how it got like that, but your local checkout was in a bad state. That target forced all the submodules back into a good state and through away all the build artifacts restoring you to the state of a clean checkout. Master is the latest development version of PX4 that you get by default when you. If was a secondary suggestion in case you were on an old version.

closed 2017-05-12 by dagar