PX4/PX4-Autopilot #7231: STABILIZED to AUTO.LAND drops actuator outputs to min PWM

lamping7, 2017-05-13:
Upon triggering AUTO.LAND from STABILIZED, the FC drops all actuator outputs to min PWM, causing instability before re-applying thrust to regain altitude.

Log: http://review.px4.io/plot_app?log=61623410-7cb8-401d-8a95-9f08a6c91440

LorenzMeier, 2017-05-13:
I've looked into this and your log is oddly incomplete. I will need to dig a little deeper into it.

LorenzMeier, 2017-05-20:
Can you re-test this with RC3? The log shows big errors on the estimator side but is too incomplete to judge what the issue is or if this is just the log.

Try also to replace the microSD card with a better model / brand.

lamping7, 2017-05-31:
I will not be able to test for a few weeks.

lamping7, 2018-01-30:
I don't recall this happening in the most recent flights. I'll re-open if I see this when flying resumes in the Spring.

closed 2018-01-30 by lamping7