PX4/PX4-Autopilot #7233: [Optical Flow]Can't estimate local position by EKF2

timfaner, 2017-05-13:
I used optical flow without gps but local position from pixhawk board is drifting and I have checked my optical flow module ,it  works well.
I have set EKF_AID_MASK to 2 and EKF_HGT_MODE to Range sensor.
Is their any problem I did wrong? Please help me..
Thank you.

LorenzMeier, 2017-05-13:
@ChristophTobler Can you comment?

timfaner, 2017-05-14:
Here's my flight log [log](http://logs.px4.io/plot_app?log=ec88c1f8-c5e0-4ce1-9132-cfec02db2832)
It seems local position y not controlled at all.
I control pixhawk using Mavlink command:set_position_target_local_ned and set bitmap to control it position. But it seems not work.

ChristophTobler, 2017-05-15:
@timfaner Your quad is not tuned at all http://logs.px4.io/plot_app?log=ec88c1f8-c5e0-4ce1-9132-cfec02db2832#Nav-Roll-Angle. All tuning parameters are still default which is very unlikely. You have to tune the attitude controller, else your quad won't fly well.
Flow quality seems good but your distance sensor is very noisy http://logs.px4.io/plot_app?log=ec88c1f8-c5e0-4ce1-9132-cfec02db2832#Nav-Distance-Sensor. I guess it is a sonar? These don't work very well on several grounds.

timfaner, 2017-05-17:
@ChristophTobler Thank you for reply.
I tuned quad in past two days and quad still drifts.
I placed quad still and in QGC I found LOCAL_POSITION_NED_X, LOCAL_POSITION_NED_Y drift about 0.01m/s.
Then I found  LOCAL_POSITION_NED_vx, LOCAL_POSITION_NED_vy is negative when quad still.
[image]
However Optical_flow_integrated_x , Optical_flow_integrated_y works fine.
So how can I fix drift ? Or is there anything I do wrong?
Hope your comment.

ChristophTobler, 2017-05-17:
@timfaner can you provide a log again? And maybe a video would also help.

ChristophTobler, 2017-05-17:
@timfaner You did calibrate all your sensors, right?

timfaner, 2017-05-17:
@ChristophTobler Sorry , late for reply.
Here is a [ideo](https://youtu.be/beiCyzvmYis).
I calibrate all sensors except PX4Flow , I just adjust its focal length to 2m .

ChristophTobler, 2017-05-17:
@timfaner thanks! Could you also send the log file of that flight?

ChristophTobler, 2017-05-17:
The ground you have in your flying arena is definitely not suited for optical flow as it is almost completely white and has no texture. Try again with the normal ground. To analyse it more, I would need the log file of that flight.

timfaner, 2017-05-17:
Here is the [log](http://logs.px4.io/plot_app?log=cdfea96a-c47c-445c-bc6c-4a2270acb45d).
It takes some time to find right  log.
I will try again with the normal ground tomorrow.
Thank you for advice.

ChristophTobler, 2017-05-18:
Thanks. So tuning is a bit better but could still be improved. Your mag seems to correlate with your thrust http://logs.px4.io/plot_app?log=cdfea96a-c47c-445c-bc6c-4a2270acb45d#Nav-Thrust-and-Magnetic-Field. Is your mag close to some power cable? If so, you should place it somewhere else.
Your quad is in `NAVIGATION_STATE_DESCEND` mode which is weird. It should be in position. Can you try taking off manually and then switch to altitude and then position mode?
Another thing: Have you mounted your PX4Flow the correct way -> https://dev.px4.io/en/tutorials/optical_flow.html?

timfaner, 2017-05-22:
I set quad mode into offboard in order to takeoff.Yestarday  I try to taking off manually and then switch to altitude and then position mode. When switched to altitude quad just go up . [Here is log.](http://logs.px4.io/plot_app?log=91cba99c-45f3-4645-8c6f-56cc4403541f). I think I might ignore some parameters and quad can't fly normally. Can you please tell me paraments to be set and step should be flow? Thank you...

ChristophTobler, 2017-05-31:
Your quad is still not tuned properly http://logs.px4.io/plot_app?log=91cba99c-45f3-4645-8c6f-56cc4403541f#Nav-Roll-Angular-Rate and you should also try dampening your autopilot differently as you have a lot of vibrations http://logs.px4.io/plot_app?log=91cba99c-45f3-4645-8c6f-56cc4403541f#Nav-Raw-Acceleration

ChristophTobler, 2017-06-06:
Please reopen if still an issue

closed 2017-06-06 by ChristophTobler