PX4/PX4-Autopilot #7234: Inspect v1.6 flight log

LorenzMeier, 2017-05-13:
http://logs.px4.io/plot_app?log=727281fb-388a-4b9f-ae96-65bf88ddd1ce

santiago3dr, 2017-05-16:
after troubleshooting said copter, I am led to believe the radio failsafe was not set properly on the FrSky receiver and when connection was lost, all inputs were set to minimum value which led to the fall and crash.

dagar, 2017-05-17:
Resolved?

santiago3dr, 2017-05-17:
i believe so; can you confirm log shows typical rc loss given input channels go to 0?

dagar, 2017-05-17:
rc_signal loss is 0 the entire log.

[image]

d3z0rian, 2017-05-17:
I think is expected to see rc_signal loss = 0, since the receiver wasn't
configured properly, so even when there is no communication with the RC
transmitter, the receiver will send RC signals to the autopilot.

We'll be more careful to check the failsafe is in place before we fly.

closed 2017-05-18 by LorenzMeier