PX4/PX4-Autopilot #7237: MC: Max Alt of 100m in PosCTL

tops4u, 2017-05-14:
I'm currently flying some trainings Missions for SAR with Dogs. I experienced a serious Problem that in PosCTL Mode I'm not able to climb above 100m AGL. Stabilized allows this. I tried various settings and disabled Geofencing. As Logupload on logs.px4.io throws a 500 Error, I'll attach the log directly. 

I can not find any other limiting parameter and stabilized allows going higher - but this is currently no option. 
FW git-hash: 6c6cfd4f65adeb0cc79611f140f99db2aead5dfb

[log_139_2017-5-14-10-41-04.ulg.zip](https://github.com/PX4/Firmware/files/999203/log_139_2017-5-14-10-41-04.ulg.zip)

LorenzMeier, 2017-05-14:
Fixed in https://github.com/PX4/Firmware/commit/cdb8146af5c47a13ea0bfc03c65b62786c4a2043. Needs further cleanup, but with this change you should not run into it.

closed 2017-05-14 by LorenzMeier

tops4u, 2017-05-14:
@LorenzMeier thanks for this super fast fix. I have seen this parameter but it was unclear what this meant and I ignored it as it is in the Land Group - a function that I don't use at all. 
We had clearance to up to 500m for this Training.

For it is still a bit confusing why this applies to PosCtl but not Stab FlightMode.