PX4/PX4-Autopilot #7239: navio_gpio only works on RPI2

nicolaerosia, 2017-05-14:
navio_gpio has the Physical Memory offset hardcoded, https://github.com/PX4/Firmware/blob/master/src/drivers/navio_gpio/navio_gpio.h#L38 , which only works for RPi2, on RPi3 it hangs.

Is there a reason why we want to write directly to memory instead of using the sysfs-based interface for GPIOs, http://elinux.org/RPi_GPIO_Code_Samples#sysfs ?
I assume there is no user of navio_gpio who expects it to be a high performance interface.
If you agree, I could create a Linux gpio sysfs driver, replacing navio_gpio, and then convert navio_rgbled to use that.

mhkabir, 2017-05-14:
Sounds good!

nicolaerosia, 2017-05-15:
Great, I've created a PR for this

nicolaerosia, 2017-05-29:
PR merged

closed 2017-05-29 by nicolaerosia