PX4/PX4-Autopilot #7240: Linux watchdog

nicolaerosia, 2017-05-14:
What are you using to automatically restart px4 in case of Segmentation fault on Linux?
I was thinking of using systemd.

mhkabir, 2017-05-14:
If you have a segfault in the first place, you're already falling out of the sky. The idea would be to avoid that happening at all.
But yes - systemd is a good option for managing the process.

nicolaerosia, 2017-05-15:
Does px4 support to be started in mid air? I believe you can't expect to have a software that will never fail. For instance, read problem 3 from https://github.com/PX4/Firmware/issues/7223 :
```
mavlink streams linked list: the receiver thread iterates this list (for reading the message interval), and it's possible that the main thread deletes an item from the list (when setting the rate to 0) at the same time leading to invalid memory access.
```
Most likely, on Linux, this will trigger a segfault.
If PX4 crashes on Linux, PWM will still be generated since that is controlled using sysfs and does not have a watchdog. A systemd service which stops PWM after px4 crashes can avoid this.

dagar, 2017-05-17:
It would be nice to expand the details here. I've never even used px4 on linux outside of the build environment. CMake should generate the packages including any appropriate init. Then we could start distributing the packages.
PX4 SITL as a desktop linux package might also be nice. @jgoppert

closed 2018-02-21 by PX4BuildBot