PX4/PX4-Autopilot #7243: Max logging size for LOGGER

western-co-at, 2017-05-15:
When running the logger over a long time with a 8GBytes SD card I noticed that the reported size of the log file skipped at 4GB to back to zero because of 32 bit fat limit. When logging e.g. 6GBytes the reported size under Windows of the log file is 2GBytes but the used size of the SD card is 6GBytes. After deleting the log file, the SD card releases the whole memory of the log file again (in this case 6GByte). Would it make sense to split the log file into a further file after a file size of 4GByte?

julianoes, 2017-05-15:
FYI @bkueng

bkueng, 2017-05-15:
@davids5 IMO NuttX should handle this, and `write` should return an error when trying to go beyond the file system limit (4GB is not necessarily a hard limit, e.g. on Linux with Ext FS, it's not a problem to write more than that). Seems to me that NuttX does not handle this correctly?

@western-co-at do you have a valid use-case for this or is it just about handling the failure case properly? With 4GB you can log about a full day.

davids5, 2017-05-15:
@bkueng - I will look into it.

western-co-at, 2017-05-16:
@bkueng no I have no use case for this. I'm aware that it take quite a lot of time till the 4GB limit is reached.

closed 2018-02-21 by PX4BuildBot