PX4/PX4-Autopilot #7251: Quad using flow lands, when enabling GPS and being in takeoff (auto)

ChristophTobler, 2017-05-16:


ChristophTobler, 2017-06-07:
This is fixed once we use local instead of global coordinates inside navigator/pos_controller -> https://github.com/PX4/Firmware/tree/global_to_local (WIP)

mhkabir, 2017-06-07:
@ChristophTobler @Stifael Do you think this is a good opportunity to run through the system and make sure everything operates in the local frame? We should just project whatever global stuff we need (e.g waypoints) into the local frame.

ChristophTobler, 2017-06-07:
@mhkabir Yes, that's our goal. Navigator and ms_pos_controller already use local coordinates and it works with EKF2 but needs some more testing. I just saw that we need to check with the LPE as we still have some issues there. Could you maybe help us out, since you are familiar with the LPE?

mhkabir, 2017-06-07:
Yeah I can have a look, sure. Can you point me to where we are having issues?

I think it would also be great if you guys could quickly look at this and we discuss about how we can bring it with least-painful rebases :) https://github.com/PX4/Firmware/pull/7363.

ChristophTobler, 2017-06-07:
Sure, I'll have a look at it.

https://github.com/PX4/Firmware/pull/7370 seems to have fixed the LPE issues. But I guess it would still help if you could have a general look at it and test it :)

ChristophTobler, 2017-09-18:
Actually, now the quad goes sideways as well...
To reproduce the results run `make posix_sitl_default gazebo_iris_opt_flow` followed by a `commander takeoff` and then set the `EKF2_AID_MASK` to 3 (enable GPS as well).

priseborough, 2017-09-19:
When the EKF sets the origin for the local position in-flight, the position controller is changing the set-point by the amount of the change in origin for 0,0. The origin change time stamp is being set correctly at the time of the origin set and  prior to the origin set, xy_global and z_global are output as false by the EKF.

The position controller does not appear to be checking the validity of the local position origin via the xy_global and z_global messages.

priseborough, 2017-09-19:
The source of the problem is in MulticopterPositionControl::update_ref() and appears to be fixed with the following minor changes: https://github.com/PX4/Firmware/tree/pr-fixPosCtrlBug

ChristophTobler, 2017-09-19:
@priseborough Nice! Works in my simulation.

ChristophTobler, 2017-09-19:
https://github.com/PX4/Firmware/pull/7983

ChristophTobler, 2017-09-25:
Fixed in https://github.com/PX4/Firmware/pull/7983

closed 2017-09-25 by ChristophTobler