PX4/PX4-Autopilot #7253: RPI Preflight checks are disabled

nicolaerosia, 2017-05-16:
is there a particular reason why preflight checks are disabled on RPI ?
https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L563
I've used git blame to track where it cames from, https://github.com/PX4/Firmware/commit/76ee17e532fdbd28bc6e0a20bf16626c17f52288 , but it provides no reason.

dagar, 2017-05-16:
My guess would be that corresponds with bringing it up to the point it's flyable and hitting problems. @bkueng?

Do you want to re-enable them and start opening or working through the issues?

bkueng, 2017-05-17:
Yes it was to simplify bring-up. I don't know if there are remaining issues, but we should push towards enabling them.

nicolaerosia, 2017-05-18:
I'll enable them and see what happens

nicolaerosia, 2017-05-18:
I have calibrated the sensors, commander is reporting that there are not sensors:

```
WARN  [commander] PREFLIGHT FAIL: NO MAG SENSOR #0
WARN  [commander] Warning: Primary compass not found
WARN  [commander] PREFLIGHT FAIL: NO ACCEL SENSOR #0
WARN  [commander] Warning: Primary accelerometer not found
WARN  [commander] PREFLIGHT FAIL: NO GYRO SENSOR #0
WARN  [commander] Warning: Primary gyro not found
```

Here are the devices:
```
pxh> list_devices
INFO  [drivers__device] PX4 Devices:
INFO  [drivers__device] DF Devices:
INFO  [drivers__device]    /dev/imu0
INFO  [drivers__device]    /dev/imu1
INFO  [drivers__device]    /dev/imu2
INFO  [drivers__device]    /dev/baro0
INFO  [drivers__device]    /dev/rgbled0
INFO  [drivers__device]    /dev/adc0
pxh> list_files
INFO  [drivers__device] Files:
pxh> list_topics
INFO  [drivers__device] Devices:
INFO  [drivers__device]    /obj/_obj_
INFO  [drivers__device]    /obj/log_message0
INFO  [drivers__device]    /obj/parameter_update0
INFO  [drivers__device]    /obj/sensor_accel0
INFO  [drivers__device]    /obj/sensor_gyro0
INFO  [drivers__device]    /obj/sensor_mag0
INFO  [drivers__device]    /obj/sensor_accel1
INFO  [drivers__device]    /obj/sensor_gyro1
INFO  [drivers__device]    /obj/sensor_mag1
INFO  [drivers__device]    /obj/gps_inject_data0
INFO  [drivers__device]    /obj/input_rc0
INFO  [drivers__device]    /obj/rc_parameter_map0
INFO  [drivers__device]    /obj/differential_pressure0
INFO  [drivers__device]    /obj/vehicle_control_mode0
INFO  [drivers__device]    /obj/actuator_controls_00
INFO  [drivers__device]    /obj/sensor_correction0
INFO  [drivers__device]    /obj/sensor_selection0
INFO  [drivers__device]    /obj/sensor_combined0
INFO  [drivers__device]    /obj/sensor_preflight0
INFO  [drivers__device]    /obj/led_control0
INFO  [drivers__device]    /obj/vehicle_status0
INFO  [drivers__device]    /obj/actuator_armed0
INFO  [drivers__device]    /obj/safety0
INFO  [drivers__device]    /obj/mission_result0
INFO  [drivers__device]    /obj/geofence_result0
INFO  [drivers__device]    /obj/manual_control_setpoint0
INFO  [drivers__device]    /obj/offboard_control_mode0
INFO  [drivers__device]    /obj/vehicle_global_position0
INFO  [drivers__device]    /obj/vehicle_local_position0
INFO  [drivers__device]    /obj/vehicle_attitude0
INFO  [drivers__device]    /obj/vehicle_land_detected0
INFO  [drivers__device]    /obj/vehicle_gps_position0
INFO  [drivers__device]    /obj/vehicle_command0
INFO  [drivers__device]    /obj/battery_status0
INFO  [drivers__device]    /obj/subsystem_info0
INFO  [drivers__device]    /obj/position_setpoint_triplet0
INFO  [drivers__device]    /obj/system_power0
INFO  [drivers__device]    /obj/vtol_vehicle_status0
INFO  [drivers__device]    /obj/cpuload0
INFO  [drivers__device]    /obj/fw_pos_ctrl_status0
INFO  [drivers__device]    /obj/home_position0
INFO  [drivers__device]    /obj/onboard_mission0
INFO  [drivers__device]    /obj/offboard_mission0
INFO  [drivers__device]    /obj/airspeed0
INFO  [drivers__device]    /obj/optical_flow0
INFO  [drivers__device]    /obj/distance_sensor0
INFO  [drivers__device]    /obj/vehicle_vision_position0
INFO  [drivers__device]    /obj/vehicle_vision_attitude0
INFO  [drivers__device]    /obj/control_state0
INFO  [drivers__device]    /obj/ekf2_timestamps0
INFO  [drivers__device]    /obj/vehicle_attitude_setpoint0
INFO  [drivers__device]    /obj/vehicle_rates_setpoint0
INFO  [drivers__device]    /obj/multirotor_motor_limits0
INFO  [drivers__device]    /obj/vehicle_command_ack0
INFO  [drivers__device]    /obj/sensor_baro0
INFO  [drivers__device]    /obj/actuator_outputs0
INFO  [drivers__device]    /obj/telemetry_status0
INFO  [drivers__device]    /obj/commander_state0
INFO  [drivers__device]    /obj/satellite_info0
INFO  [drivers__device]    /obj/actuator_controls_10
INFO  [drivers__device]    /obj/vehicle_local_position_setpoint0
INFO  [drivers__device]    /obj/vehicle_global_velocity_setpoint0
INFO  [drivers__device]    /obj/att_pos_mocap0
INFO  [drivers__device]    /obj/rc_channels0
INFO  [drivers__device]    /obj/esc_status0
INFO  [drivers__device]    /obj/estimator_status0
INFO  [drivers__device]    /obj/ekf2_innovations0
INFO  [drivers__device]    /obj/tecs_status0
INFO  [drivers__device]    /obj/wind_estimate0
INFO  [drivers__device]    /obj/camera_trigger0
INFO  [drivers__device]    /obj/camera_capture0
INFO  [drivers__device]    /obj/gps_dump0
INFO  [drivers__device]    /obj/task_stack_info0
INFO  [drivers__device]    /obj/uavcan_parameter_value0
INFO  [drivers__device]    /obj/vehicle_status_flags0
INFO  [drivers__device]    /obj/mc_att_ctrl_status0
INFO  [drivers__device]    /obj/mavlink_log0
INFO  [drivers__device]    /obj/uavcan_parameter_request0
```

As you can see, there's no /dev/gyroX (GYRO_BASE_DEVICE_PATH) which commander tries to open, but there is /obj/sensor_gyro0.
Any pointers are appreciated

LorenzMeier, 2017-05-18:
We need to refactor these checks to use the uORB API instead

dagar, 2017-05-18:
Options
1. Split /dev/imu into /dev/gyro, accel, mag (DriverFramework)
2. Update the calibration code to be able to read combined /dev/imu
3. Update the calibration code to read from uORB instead of the block device.

nicolaerosia, 2017-05-18:
I have created a PR which enables some of the checks.
Do you have in plan to do the refactoring?

closed 2019-02-13 by stale[bot]