PX4/PX4-Autopilot #7258: Release v1.6 sign-off

mrpollo, 2017-05-17:
Is your team ready for a release?

* [ ] Project / Founder - @LorenzMeier 
* [ ] Dev Call - @mrpollo
* [ ] Communication Architecture - @bkueng , @julianoes 
* [ ] UI / UX - @dogmaphobic 
* [ ] Multicopter Flight Control - @Stifael, @MaEtUgR
* [ ] VTOL Flight Control - @tumbili, @AndreasAntener, @sanderux 
* [ ] Fixed Wing Flight Control - @dagar, @priseborough 
* [ ] Racers - @anton-matosov 
* [ ] OS / drivers - @davids5 
* [ ] UAVCAN / Industrial - @pavel-kirienko 
* [ ] State Estimation - @jgoppert, @priseborough 
* [ ] VIO - @mhkabir, @ChristophTobler 
* [ ] Obstacle Avoidance - @vilhjalmur89 
* [ ] Snapdragon - @mcharleb 
* [ ] Intel Aero - @lucasdemarchi, @simonegu 
* [ ] Raspberry Pi / Navio - @bkueng 
* [ ] Parrot Bebop - @eyeam3 
* [ ] Airmind MindPX / MindRacer - @iZhangHui 
* [ ] Test team - @d3z0rian, @mrpollo

jgoppert, 2017-05-17:
@priseborough how is EKF2 doing. I think LPE is stable.

davids5, 2017-05-17:
There was a fix in upstream Nuttx this morning stm32_serial: fix freezing serial port I will need to back port. ASAP.

 

David

dagar, 2017-05-17:
Could someone comment on the state of Pixhawk 3 support in v1.6.0? @klopezal, @sanderux

mhkabir, 2017-05-17:
@dagar It is supported fully. We are waiting for fresh/fixed v1.3 hardware before we can polish the experience more. We are also extensively test flying it on a VTOL config.

dagar, 2017-05-17:
Thanks @mhkabir. Could you do a quick pass and see if any of these are still relevant? https://github.com/PX4/Firmware/issues?q=is%3Aopen+is%3Aissue+label%3Apixhawk3

mhkabir, 2017-05-17:
All of those issues have hardware fixes - but we don't have latest hardware yet, so they're still all relevant, yes.

eyeam3, 2017-05-17:
PX4 on the Parrot Bebop is not running properly at the moment, but it should not gate the release.

priseborough, 2017-05-17:
EKF2 has a PR pending for miscellaneous updates https://github.com/PX4/Firmware/pull/7252. 

Items 5 and 6 close vulnerabilities, and item 1 adds an important logging message for diagnostics, the others enable a 'nice to have' feature . I can split the 'nice to have' into a separate PR if required.

davids5, 2017-05-17:
>There was a fix in upstream Nuttx this morning stm32_serial: fix freezing serial port I will need to back port. ASAP.

PR is https://github.com/PX4/Firmware/pull/7260

priseborough, 2017-05-17:
I have closed the pending PR and split it into the robustness improvements that I think should go into the release:

https://github.com/PX4/Firmware/pull/7265

And the nice to have:

https://github.com/PX4/Firmware/pull/7264

bkueng, 2017-05-18:
I think this one should go in: https://github.com/PX4/Firmware/pull/7226

LorenzMeier, 2017-05-18:
I flagged all PRs with the milestone that I think should still go in.
https://github.com/PX4/Firmware/milestone/22

nicolaerosia, 2017-05-18:
navio_gpio is broken on RPI3, see https://github.com/PX4/Firmware/pull/7241

DonLakeFlyer, 2017-05-18:
FYI: I don't have time till tomorrow, then I'll do some SITL runs and sign off.

sanderux, 2017-05-19:
@LorenzMeier should those pr's not go in before we test / sign off?

nicolaerosia, 2017-05-19:
Maybe this should also go in https://github.com/PX4/Firmware/pull/7272

LorenzMeier, 2017-05-26:
I'm closing this as we discussed this on the last dev call.

closed 2017-05-26 by LorenzMeier