PX4/PX4-Autopilot #7259: log CPU and RAM per process

dagar, 2017-05-17:
@bkueng do you think it would be realistic to log this per process? Given that's it difficult to impossible to profile NuttX it would be extremely helpful for understanding the code from a resource perspective. 

Then display as a stacked line plot on flight review.

bkueng, 2017-05-18:
We currently do that twice, preflight and postflight (together with the perf counters). You can use `ulog_info -v <file>` to display the data. Flight review does not display it yet.

Keeping an eye on resource usage is important, but I want to avoid adding too much instrumentation-overhead in general. CPU-load is typically very stable during flight, up to certain events like VTOL transitions or param changes. So I think the current state is enough for now. Even if we logged this more frequently, it would not replace profiling, as it would not be possible to find things outlined in https://github.com/PX4/Firmware/pull/6829 for example.
With the addition of the mavlink shell to QGC, it also becomes much easier to monitor specific things in-flight.

dagar, 2017-05-18:
Insight into things like VTOL transition is exactly what I was thinking of.

I'd still like a NuttX profiling option that's usable during HIL, but so far haven't found anything.
Maybe NuttX and QEMU? Probably a none trivial undertaking.

closed 2017-05-30 by dagar