PX4/PX4-Autopilot #7261: Air frame init script files attempt to set rates that will fails on all FMU

davids5, 2017-05-17:
There are 2 issue:
1. Air frame init script files attempt to set rates that will fails on all FMU
2. Need a way to set the default rate uses when an alt rates is applied.

## Issue 1:
Air frame script 
```set PWM_OUT 5```

Later command
```pwm rate -c ${PWM_OUT} -r ${PWM_RATE}```

FMUv2 HW: default is /dev/pwm_output0
```
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
```
2) FMUv4 HW: default is /dev/pwm_output0
```
channel group 0: channels 1 2 3 4
channel group 1: channels 5 6
```

FMUv2
```
nsh>  pwm rate -c 5 -r 400
ERROR [pwm] PWM_SERVO_SET_SELECT_UPDATE_RATE
```
It would have to be ```pwm rate -c 5678 -r 400``` On V2

FMUv4
```
pwm rate -c 5 -r 400
WARN  [fmu] rate group 1 mask 30 bad overlap 10
ERROR [pwm] PWM_SERVO_SET_SELECT_UPDATE_RATE
```
It would have to be ```pwm rate -c 56 -r 400``` On V4

## Issue 2:
Assuming FMUv2 or FMUv4
```pwm rate -g 0 -r 400``` Sets group 0 to ``alternative``` rate of 400 Hz and ALL other groups to ```default rate``` 50 Hz
```pwm rate -g 1 -r 25``` Sets group 1 to ``alternative``` rate of 25 Hz and ALL other groups to ```default rate``` 50. Thus undoing the Group 0 rate of 400.

dagar, 2017-05-18:
I'm not sure I completely understand all the details here, but here are a few thoughts.

1. Most PX4 developers seem to be completely unaware of the single rate per group limitation. When this fails the error needs to be visible.
2. Users load the same airframes on both FMU+IO (px4fmu-v2, v3, v4pro, etc) devices and FMU only devices (pixracer, mindracer). If we rearrange the IO groups to 1234, 56, 78, then we can at least be sure that any airframe that can be loaded on the pixracer will also work on a pixhawk.
3. We need to prevent airframes from being loaded on inappropriate targets.
4. We need to do a pass and make sure all airframes are still valid in general, and especially with respect to 2) and 3).
5. PWM setup is confusing for mixed configurations (ESCs + motors) and the single params (PWM_RATE, PWM_MIN, PWM_MAX, PWM_DISARMED) are only used at startup via init script. Is there a better way to do this?

closed 2018-02-21 by PX4BuildBot

feemi, 2019-12-17:
Hi @dagar  Is there any progress on this issue? we are experiencing issues on vtol configuration. 
Our mixer is similar to delta quad which uses only main pwm outputs. But its not possible to set different rates on pusher and elevons seperately.