PX4/PX4-Autopilot #7262: Bug?  RTL and Pause locked out until takeoff waypoint achieved.

Antiheavy, 2017-05-17:
@DonLakeFlyer suggested I post this here for @LorenzMeier and others.

We had a flight on our fixed wing this week where RTL and Pause locked out (grayed out) in Qgc until takeoff waypoint was achieved.  It should be available as soon as flight is detected I believe. 

@DonLakeFlyer says:
"Firmware bug with respect to EXTENDED_SYS_STATE.landed_state mavlink message values which shows up when you replay this log:
    Vehicle armed: landed_state = 1, MAV_LANDED_STATE_ON_GROUND
    Vehicle takes off: landed_state = 3, MAV_LANDED_STATE_ON_GROUND + MAV_LANDED_STATE_IN_AIR
    It eventually clears back to landed_state = 2, MAV_LANDED_STATE_IN_AIR
The combination of MAV_LANDED_STATE_ON_GROUND + MAV_LANDED_STATE_IN_AIR should never happen and confuses QGC hence the buttons get all out of whack."

Qgc .tlog attached.
[2017-05-16.18-00-30.zip](https://github.com/PX4/Firmware/files/1009314/2017-05-16.18-00-30.zip)

PX4 log file link:
http://review.px4.io/plot_app?log=d7b1cfd2-c1c2-4410-a4cf-1691bf922004

Note: there is a also a problem with EKF2 being unhappy about life during climbout and then again later in the flight.  This is why I was hunting for the RTL button during climb-out and discovered they were grayed out.  We are looking into this EKF2 issue separately @dagar .

dagar, 2017-05-18:
PX4 has a land detector (landed or not), but for takeoff and landing there's only the internal position setpoint type.

    EXTENDED_SYS_STATE
    MAV_LANDED_STATE_TAKEOFF - set anytime position setpoint current is TAKEOFF
    MAV_LANDED_STATE_LANDING - set anytime position setpoint current is LAND

For a typical mission setup this goes back to TAKEOFF (MAV_CMD_NAV_TAKEOFF) being treated like a regular waypoint, but with special vehicle behavior.

@DonLakeFlyer - do you think we could update the mavlink spec for MAV_CMD_NAV_TAKEOFF to optionally only specify an altitude for acceptance? Or would that have to be something new.

DonLakeFlyer, 2017-05-18:
This isn't an ehancement, this is a bug. Firmware is reporting both in air and on ground at the same time.

dagar, 2017-05-18:
Sorry, I missed your details. MAV_LANDED_STATE isn't a bitmask.

[image]

DonLakeFlyer, 2017-05-18:
Ah, crap I was reading the values wrong. Duh!

@Antiheavy Back to the drawing board on QGC problem then. QGC code is correct in handling these as descreet values. Just my memory fading me when looking at logs.

closed 2017-05-30 by dagar