PX4/PX4-Autopilot #7263: I am unable to pusblish rostopics using MAVLINK with gazebo and px4..getting errors

Neels001, 2017-05-17:
I am newbie to PX4 , I have installed gazebo , and Cloned the Firmware.

Mavlink, Mavros and Firmware all have been placed under catkin src folder and built. 
Using ros and Gazebo7

I am trying to publish ros topics. But I am getting the Errors .

Steps I followed :

Terminal1 :

roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"

Terminal2 :

source Tools/setup_gazebo.bash $(pwd)$(pwd)/build_posix_sitl_default

export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo

roslaunch px4 posix_sitl.launch

Errors which I am getting : 

Terminal1 : where mavros is running

[ WARN] [1493507102.919495629]: TM: Clock skew detected (-0.019824184 s). Hard syncing clocks.
[ERROR] [1493507113.704105243]: FCU: Accel #0 fail: TOUT!
[ERROR] [1493507113.706679234]: FCU: Gyro #0 fail: TOUT!
[ WARN] [1493507113.944251371]: TM: Clock skew detected(0.012025608       s). Hard syncing clocks.
[ERROR] [1493507121.030549466]: FCU: Accel #0 fail: TOUT!
[ERROR] [1493507121.036021491]: FCU: Gyro #0 fail: TOUT!
[ERROR] [1493507121.630152661]: FCU: Accel #0 fail: TOUT!
[ERROR] [1493507121.634544822]: FCU: Gyro #0 fail: TOUT!
[ERROR] [1493507122.108005687]: FCU: Accel #0 fail: TOUT!
[ERROR] [1493507122.111951530]: FCU: Gyro #0 fail: TOUT!
[ERROR] [1493507122.829282656]: FCU: Accel #0 fail: TOUT!
[ERROR] [1493507122.830428095]: FCU: Gyro #0 fail: TOUT!
[ WARN] [1493507141.422615369]: TM: Clock skew detected (-0.016021096 s). Hard syncing clocks.

Terminal 2 :  where PX4 is running .
WARN [mavlink] [timesync] Hard setting offset.
WARN [mavlink] [timesync] Hard setting offset.
ERROR [sensors] Accel #0 fail: TOUT!
ERROR [sensors] Gyro #0 fail: TOUT!
WARN [mavlink] [timesync] Hard setting offset.
WARN [mavlink] [timesync] Hard setting offset.
WARN [mavlink] [timesync] Hard setting offset.
WARN [mavlink] [timesync] Hard setting offset.
WARN [mavlink] [timesync] Hard setting offset.

I am struggling with this issue from very long time . Please kindly help me with this issue.

Thank you!!

Neels001, 2017-06-05:
Solved .. Upgrading RAM , adding proprietary drivers for graphics and following Offboard example given in PX4 documentation.

closed 2017-06-05 by Neels001

Kailash-Rangaraj, 2017-12-19:
Hi @Neels001 I am facing the same issue. I installed Gazebo simulator and run this code "make posix gazebo_iris_opt_flow". I got the same error like yours. The errors are related to the sensors. I already changed the proprietary drivers for graphics and I wanted to know whether we need to install QGroundStation to rectify this error

YCH188, 2018-02-04:
@Neels001 @Kailash-Rangaraj I have the same error ERROR [mavlink] [timesync] Hard setting offset.
And what is that means. How can I solve the error.Thanks!