PX4/PX4-Autopilot #7266: [navigator] global position timeout in lpe, quad starts drifting away

SujithSizon, 2017-05-18:
Trying to run https://github.com/TEAMIFOR/ifor_offb2/blob/master/src/test_node.cpp (makes quad move in a circle in offboard mode). It works well till halfway but then global position time-out occurs. 

Parameters:
uorb start
param load
param set MAV_TYPE 2
param set SYS_AUTOSTART 4010
param set SYS_RESTART_TYPE 2
param set SYS_MC_EST_GROUP 1
dataman start
param set BAT_N_CELLS 3
param set CAL_GYRO0_ID 2293768
param set CAL_ACC0_ID 1376264
param set CAL_ACC1_ID 1310728
param set CAL_MAG0_ID 196616
param set CAL_GYRO0_XOFF 0.01
param set CAL_ACC0_XOFF 0.01
param set CAL_ACC0_YOFF -0.01
param set CAL_ACC0_ZOFF 0.01
param set CAL_ACC0_XSCALE 1.01
param set CAL_ACC0_YSCALE 1.01
param set CAL_ACC0_ZSCALE 1.01
param set CAL_ACC1_XOFF 0.01
param set CAL_MAG0_XOFF 0.01
param set SENS_BOARD_ROT 0
param set SENS_BOARD_X_OFF 0.000001
param set COM_RC_IN_MODE 1
param set NAV_DLL_ACT 2
param set COM_DISARM_LAND 3
param set NAV_ACC_RAD 2.0
param set RTL_RETURN_ALT 30.0
param set RTL_DESCEND_ALT 10.0
param set RTL_LAND_DELAY 0
param set MIS_TAKEOFF_ALT 2.5
param set MC_ROLLRATE_P 0.3
param set MC_PITCHRATE_P 0.3
param set MC_PITCH_P 5.5
param set MC_ROLL_P 5.5
param set MC_ROLLRATE_I 0.1
param set MC_PITCHRATE_I 0.1
param set MPC_HOLD_MAX_Z 2.0
param set MPC_Z_VEL_P 0.8
param set MPC_Z_VEL_I 0.15
param set MPC_XY_VEL_P 0.15
param set MPC_XY_VEL_I 0.2

param set MC_PITCH_P 5.0
param set MC_ROLL_P 5.0
param set MC_ROLLRATE_P 0.2
param set MC_PITCHRATE_P 0.2
param set MC_ROLLRATE_I 0.05
param set MC_PITCHRATE_I 0.05
#param set LPE_GPS_ON 0
param set MPC_ALT_MODE 1
param set LPE_T_MAX_GRADE 0
param set MPC_XY_VEL_MAX 2
param set MPC_Z_VEL_MAX_DN 2
param set MPC_TILTMAX_AIR 10
param set MPC_TILTMAX_LND 10
param set MPC_XY_P 0.5
param set MIS_TAKEOFF_ALT 2
param set NAV_ACC_RAD 1.0
param set CBRK_GPSFAIL 240024

param set LPE_FUSION 242
param set LPE_FAKE_ORIGIN 1

replay tryapplyparams
simulator start -s
tone_alarm start
gyrosim start
accelsim start
barosim start
adcsim start
pwm_out_sim mode_pwm
sensors start
commander start
land_detector start multicopter
navigator start
attitude_estimator_q start
local_position_estimator start
mc_pos_control start
mc_att_control start
mixer load /dev/pwm_output0 ROMFS/px4fmu_common/mixers/quad_dc.main.mix
mavlink start -u 14556 -r 2000000
mavlink start -u 14557 -r 2000000 -m onboard -o 14540
mavlink stream -r 80 -s POSITION_TARGET_LOCAL_NED -u 14556
mavlink stream -r 80 -s LOCAL_POSITION_NED -u 14556
mavlink stream -r 80 -s GLOBAL_POSITION_INT -u 14556
mavlink stream -r 80 -s ATTITUDE -u 14556
mavlink stream -r 80 -s ATTITUDE_QUATERNION -u 14556
mavlink stream -r 80 -s ATTITUDE_TARGET -u 14556
mavlink stream -r 20 -s RC_CHANNELS -u 14556
mavlink stream -r 250 -s HIGHRES_IMU -u 14556
mavlink stream -r 10 -s OPTICAL_FLOW_RAD -u 14556
slogger start -e -t
mavlink boot_complete
replay trystart

px4 log file:  http://logs.px4.io/plot_app?log=7085ce9d-ef84-4464-a774-7e7db81b8186

INFO  [simulator] Got initial simulation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
INFO  [tone_alarm] startup
[lpe] fuse gps: 0, flow: 1, vis_pos: 0, vis_yaw: 0, land: 1, pub_agl_z: 1, flow_gyro: 1
INFO  [mavlink] mode: Normal, data rate: 2000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 2000000 B/s on udp port 14557 remote port 14540
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
pxh> INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2017-05-18/06_28_54.ulg
INFO  [local_position_estimator] [lpe] Lidar detected with ID 0
INFO  [local_position_estimator] [lpe] land init
INFO  [local_position_estimator] [lpe] global origin init (parameter) : lat  47.40 lon   8.55 alt   0.0 m
INFO  [local_position_estimator] [lpe] lidar init: mean 9 cm stddev 0 cm
INFO  [local_position_estimator] [lpe] baro init 0 m std 7 cm
INFO  [commander] home: 47.3977432, 8.5455942, 0.15
INFO  [tone_alarm] home_set
INFO  [mavlink] partner IP: 127.0.0.1
WARN  [mavlink] [timesync] Hard setting offset.
WARN  [mavlink] [timesync] Hard setting offset.
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977433, 8.5455944, 0.10
INFO  [tone_alarm] positive
INFO  [commander] home: 47.3977433, 8.5455944, 0.10
INFO  [tone_alarm] arming
INFO  [commander] Takeoff detected
Number of good matches: 4, desired: 20
INFO  [local_position_estimator] [lpe] land timeout 
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/gazebo/command/motor_speed, deleting message. This warning is printed only once.
WARN  [navigator] global position timeout
WARN  [navigator] global position timeout

dagar, 2017-05-18:
@jgoppert could you take a look?

dagar, 2017-05-18:
Why does home init twice?

SujithSizon, 2017-05-18:
i dont know, whenever i try to arm using qgc this happens

TSC21, 2017-07-27:
@SujithSizon updates?

closed 2018-02-21 by PX4BuildBot