PX4/PX4-Autopilot #7270: Multiple Vehicle Simulation PX4 SITL for Planes

paan2097, 2017-05-18:
There are multiple issues floating around for multi uav px4 sitl simulations but I'm having trouble getting any of them working.....there is a need for a stable and clear approach for running multiple vehicles in SITL all connected to gazebo. The multiple approaches only support iris model, but there needs to be support for all vehicles, especially planes in my case. Please provide clear instructions on the dev guide on how to run multiple vehicles in SITL connected to gazebo and support for not only iris, but planes. Please help @LorenzMeier 

Here's this issues I've been following:
https://github.com/PX4/Firmware/issues/6408
https://github.com/PX4/Firmware/pull/2869
https://github.com/PX4/Firmware/pull/7235
https://github.com/PX4/Firmware/issues/6699

Some separate repositories offering their own solution, but only for iris:
https://github.com/dennisss/tansa/tree/master/doc/examples/gazebo_spawn
https://github.com/acsl-mipt/multiple-sitl

dagar, 2017-05-18:
This might be a good excuse to refactor the convoluted SITL launch.

paan2097, 2017-05-30:
@dagar Any hope for refactoring soon? I've adapted denniss/tansa repo to work for multiple planes but am still running into multiple issues. Thanks

jgoppert, 2017-05-30:
I did some work on this. I wrote a modified vehicle spawn python script where you can specify sdf params to rewrite. I will submit it as a pr when I get it cleaned up.

dagar, 2017-05-30:
I'm interested in helping, but I haven't actually needed multi vehicle sim yet. Someone more familiar with gazebo and in need of the solution should probably drive the effort. With the design/architecture for handling multiple vehicles in place I'd want to help make it as clean/simple as possible for the 1 vehicle case.

closed 2018-02-21 by PX4BuildBot