PX4/PX4-Autopilot #7275: Emild reach RTK GPS support

tdnet12434, 2017-05-21:
Hi, i buy new emild reach RTK but not found any document about how it communicate to PX4.
I know that PX4 firmware doesn't support ERB format from Emild. But the NMEA should work fine?
To do this, i should connect UART from Emild reach RTK to the Serial port 4/5 on Pixhawk right? 
Because i think the standalone GPS is still needed as it built-in compass is required.
So, PX4 firmware is currently support NMEA msg format serial input or not?

Or any advice would be appreciate.

Best regard.
Thanabadee

paan2097, 2017-08-16:
Did you ever get the emlid reach RTK working with Pixhawk?

mhkabir, 2017-08-16:
Most of the needed parts are in there already. Please send a PR if you get this working :)

closed 2017-08-16 by mhkabir

andresR8, 2017-10-19:
@tdnet12434 Try to configure your emlid output with UART at 38400 of baudrate.  It worked for me.

tdnet12434, 2017-10-19:
@andresR8 Thank you for the hint!
I will try if have a chance.

paan2097, 2018-04-16:
Is this working? Is there any documentation? I can't get my emlid reach rtk working with px4.

hamishwillee, 2018-04-16:
@paan2097 There is no official documentation - we only have the ones [here](https://docs.px4.io/en/peripherals/rtk_gps_drotek_xl.html). 

@tdnet12434 @andresR8 @mhkabir Can you confirm if this worked, and what had to be done - would be great to get a page for this on the user guide.

paan2097, 2018-04-17:
@hamishwillee Thank you! I've been trying to get it up using both 1.6.0 stable firmware and then the latest 1.7.3 stable firmware. I have tried using the emlid reach's position output format via UART using NMEA and ERB formats. I've tried baud rates of 38, 57, 115. I've tried plugging in to the GPS port vs. serial 4/5 port on Pixhawk. I can't get it to work :[ Please help! Much Appreciated!

hamishwillee, 2018-04-17:
@paan2097 I can't help, that is why I have reached out to @tdnet12434 @andresR8 @mhkabir. I have also created a post on Slack in the hardware channel here: https://px4.slack.com/archives/C3GFT2QS3/p1524004278000211?thread_ts=1524004250.000370&cid=C3GFT2QS3 (generally speaking the discussion forum or Slack are good places for quick answers from the general community).

barchstien, 2019-01-25:
https://github.com/PX4/Firmware/pull/11265
PR is being reviewed, if you want to test, please go ahead, I can provide the bin to run
Once it is merged in master, it should make its way to the default distributed binary
RTK support should come soon (by February 2019 I guess)