PX4/PX4-Autopilot #7276: MC: Motors not Spinning after Arming

tops4u, 2017-05-22:
Today I had a problem, that after connecting the Battery first and then uploading a Mission via USB I was able to arm but not flying. There have been no Warnings or Errors. Also no Messages have been logged. Only Power-Cycling did resolve the Problem. 

See log (PWM Out just stays on minimum): http://logs.px4.io/plot_app?log=34d4ea13-97e0-48dc-aaf7-306c0c38e230

LorenzMeier, 2017-06-04:
I suspect its a land detector issue, yes. It might not be an issue after all but just not the right configuration. We need better documentation on that front.

deksprime, 2017-06-14:
This also happens to all of our copters from time to time, both in 1.5.5 and 1.6.0. I thought that it was somehow related to our ESC's or motors etc. Power cycle definitely solves the problem, but the reproduction of this issue is really difficult since it is completely randomized-or at least we couldn't figure out any causality with something.

tops4u, 2017-06-15:
@deksprime  I think it is highly improbable, that all ESC/Motors would fail at the same time ;-) 

Besides in the Log above it is clearly visible, that the PWM Out to the Motors stay on min PWM (1100) so the Motors don't start spinning.

LorenzMeier, 2017-06-15:
I will look with priority into this. I don't have a good handle on it yet.

tops4u, 2017-06-15:
@LorenzMeier This does not have High Prio for me as a simple Power Cycling does the trick.

LorenzMeier, 2017-06-15:
At least the log you shared has the system in position control and the throttle never reaches the default takeoff throttle setting of 65%. So I wouldn't say there is an issue in the log. If you had pushed the throttle slightly more up it would have taken off.

LorenzMeier, 2017-06-15:
Depending on how you landed before power cycling it might reset it to manual and that's why you don't see it then.

tops4u, 2017-06-15:
This was the first Flight on this day / Battery was disconnected before, I just planned and Uploaded the Mission in the trunk of my car and then placed the Vehicle on the ground.

LorenzMeier, 2017-06-15:
I'm closing this for now as resolved, but if it happens again and you are confident that you are in manual and you didn't raise throttle to 65% in position control please re-open.

closed 2017-06-15 by LorenzMeier