PX4/PX4-Autopilot #7279: OpticalFlow: POS_CTL w/o GPS

ecmnet, 2017-05-22:
Seems that POS_HOLD is rejected if using optical Flow without GPS, although local position seems to be estimated correctly. I tried it with both, FakeOrigin=1 and =0. LPE fusion flag = 242. Using LIDAR for altitude estimation (ALT_HOLD works). With vision estimates POS_HOLD works perfectly. Did I miss something here?

mhkabir, 2017-05-22:
Can you test with EKF2? We have a lot of work going on on that side, especially with respect to flow-only flight.

jgoppert, 2017-05-22:
I flew this last week or so. If we broke it in the meantime we really need to fix it before release. @ecmnet any logs would be good.

ecmnet, 2017-05-22:
@jgoppert Here's a log: http://logs.px4.io/plot_app?log=80619c3c-f179-496a-bf6d-8a31e43b84d8
@mhkabir I currently stick to LPE instead of EKF2 as my focus is currently on vision based obstacle detection and local awareness. Px4Flow in my setup is used as a positioning backup, if vision estimation fails.

TSC21, 2017-07-27:
@ecmnet updates?

ecmnet, 2017-08-17:
@TSC21 I replaced flow with RTK GPS currently as I did not get flow working properly at all (although it worked perfectly in early releases)

TSC21, 2017-08-17:
That's an expensive "upgrade". Have you reported the issue on the Flow repo?

ecmnet, 2017-08-17:
No, I tried both flow solutions (exposion branch and standard). Flow seems to work properly (at least I did not find anything wrong when connecting it directly to QGC). But I am still interested to have an indoor fallback to vision.

TSC21, 2017-08-17:
Buy what's not working then for you is the estimation?

ecmnet, 2017-08-17:
Not quite, I received some global position timeouts and pos hold was rejected, also local position was estimated. (Refer to log above). But I did not try flow since then.

MaMadDl, 2017-09-05:
i was able to Switch To POS_HOLD With Only an Optical flow algorithm that i'm working On and a sonar connected To PX4FMU_V1 Board with ADC With Modifying LPE_VXY_PUB parameter to higher value(default was 0.3 mine is 0.8) 
But When i switch to POS_HOLD, Drone always drifts Only in 1 direction. first i tried to align my direction to what is said here : [https://dev.px4.io/en/tutorials/optical_flow.html](url) . 
after that i tried to align directions to NED Frame Still not working 
here's my Log of last flight with Directions aligned on NED Frame : [https://1drv.ms/u/s!AndghYiK3yPSgv93wqk0zd2RuqRU-w](url)
The Odd Thing i noticed In middle of flight when i Switch from ALT_HOLD to POS_HOLD, Parameter LOCAL_POSITION vx and vy climbs to a random value and will not change until i land

closed 2018-02-21 by PX4BuildBot