PX4/PX4-Autopilot #7280: `mpu6000` with no other options will hard fault!

davids5, 2017-05-22:
`mpu6000` with no other options will hard fault!  

The bug is most sensors, and the complete driver code base needs a pass to 

1. use px4_get_opts
2. Use PX4 log.
3. test `argc > 1` and issue a usage warning.

mhkabir, 2017-05-22:
I can do this

davids5, 2017-05-22:
@mhkabir - 1) it might be advantageous to look at creating a helper class for use in all the drivers. The goal would be parameterising the command line parsing (just keep in mind the  need for separate data - we need it thread safe) 2) We should also look at a round up for an API using board provided bus mapping for SPI Buses/chip selects and I2C. We should have a call to discuss the second item.

bkueng, 2017-05-23:
For 1) see also https://github.com/PX4/Firmware/pull/7145

mhkabir, 2017-05-23:
@bkueng Can you handle 1) in your pull request directly? It would probably be the path of least resistance.

For 2) @davids5, let's have a call later today.

bkueng, 2017-05-23:
I'll look into it but there are many drivers and I could use some assistance. In any case I first need a solution for running multiple instances of the same driver.

closed 2018-02-21 by PX4BuildBot