PX4/PX4-Autopilot #7282: Syslink parameter acks broken Crazyflie 2

markwmuller, 2017-05-22:
Appears to be introduced by the fix for 6005, the parameter setting on the crazyflie 2's NRF are not reported correctly (they always report success, though the address appears to never be set). The immediately obvious problem is in "https://github.com/PX4/Firmware/blob/master/src/modules/syslink/syslink_main.h", where line 70 should be replaced by `int is_good(int i) { return _params_ack[i] != 0; }`. Less obvious is why setting this parameter fails.

LorenzMeier, 2017-05-23:
Thanks for reporting! @dennisss Any chance you could look into this and cross-test?

LorenzMeier, 2017-05-23:
@hamishwillee We probably should make sure the Crazyflie 2 support is documented in the user guide: https://www.bitcraze.io/crazyflie-2/ It's a really nice platform. @dennisss Can probably help.

dennisss, 2017-05-23:
I'll take a look later today and see if I can replicate it. I do remember that I think I've had this issue with address (or channel) where old versions of the NRF firmware didn't have this parameter defined and an update of the Bitcraze firmware was needed to get it working.

markwmuller, 2017-05-23:
Hi Denis -- sure, thanks. Note that the diagnostic function will now always return "good", even if the params fail (the index `i` is never used in the function `is_good`).

hamishwillee, 2017-05-24:
> @hamishwillee We probably should make sure the Crazyflie 2 support is documented in the user guide: https://www.bitcraze.io/crazyflie-2/ It's a really nice platform. @dennisss Can probably help.

@LorenzMeier I suggest we add it to the [airframes](http://px4.io/technology/airframes/) on px4.io. Should we also add [bebop](https://dev.px4.io/en/advanced/parrot_bebop.html) there? (as an aside, shouldn't [bebop](https://dev.px4.io/en/advanced/parrot_bebop.html) be in airframes section rather than debugging/advanced topics?)

I've created https://github.com/PX4/px4_user_guide/issues/19 to track adding this to the user guide. It is a little complicated because there currently isn't a good "place" for this sort of thing.

dennisss, 2017-05-24:
@markwmuller I see what you mean now. Your correction to line 70 there is definitely correct.

But, with latest master and latest NRF firmware (flashed using a crazyradio and the `cfloader flash cf2_nrf-2016.11.bin nrf51-fw` command in crazyflie-lib-python) I am able to change all three channels and connect externally via the radio. So it seems to be possibly a version issue. Otherwise, with the line patch, are you just getting a fail now in `syslink status` or is it succeeding still and it is failing to connect externally?

markwmuller, 2017-05-24:
@dennisss Yup, thanks! Updating NRF firmware now makes param set OK.

commit 60814358 (2017-07-20): Fixes #7282
closed 2017-07-20 by davids5