PX4/PX4-Autopilot #7285: the ORB_ID(vehicle_attitude_setpoint) topic in VTOL_att_control_main.cpp

garlinplus, 2017-05-23:
In this file ,the topic is not needed to publish.

dagar, 2017-05-23:
Hi @garlinplus, VTOL definitely needs to publish a vehicle_attitude_setpoint, unless there's another spot where it isn't being used. Can you point it out?

closed 2017-05-23 by dagar

garlinplus, 2017-05-24:
Hi,dager.
I search the topic in whole project, and I find four cpp file publish the topic.the four as followed:
1.fw_pos_control_l1_main.cpp
   The L1 control will modify the attitude setpoint in auto mode, so the topic is needed to publish.
2.mavlink_receive.cpp
   The gcs can send the attitude setpoint to aircraft,so the topic also should be published
3.navigator_main.cpp
   The navigator_main.cpp mainly set some mode,It doesn't need to modify attitude setpoint,.so I search in the file, and I find the publish function of  vehicle_attitude_setpoint is not called.
4.vtol_att_control_main.cpp
  I find the function of the file is to control the transition between mc mode and fix plane mode.The attitude is really controlled by mc attitude controller and plane attitude controller.So it doesn't need to publish the topic in the file.But it must subscribe the topic,because the stand quad plane need the attitude setpoint.

dagar, 2017-05-24:
VTOL is a little more complicated. It intercepts the attitude setpoint published by the FW and MC position controllers, modifies it during transitions, then republishes to the attitude controllers.