PX4/PX4-Autopilot #7288: Yaw mixing inverted during flight

bresch, 2017-05-24:
Hi everyone,

During a position control flight with my VTOL (in MC mode only), I encountered a really strange behavior. After flying perfectly well, the way response war suddenly inverted and the drone started to spin, slow down, and spin again. The sound of the vehicle showed that is was actually forcing this yaw rotation.
After analyzing the logs, it revealed that from the middle of the flight, the drone started to spin in the opposite direction of the controller output:
[image]

Indeed, until t=~58s, a positive yaw command of the yawrate controller was followed by a positive yaw response. After that time, the response is inverted.

Here are also the magnetometer innovations 
[image]

And the magnetic heading innovation
[image]

And of course, the complete [log](http://logs.px4.io/plot_app?log=55960095-7efd-4e28-8acc-1854828e75bc).

Finally, my setup is a PixHawk Pro with Orel 20 ESCs and a mRo GPS.

bresch, 2017-05-29:
New log from this morning, same issue after less than 3 minutes. 
Same sequence: 
1. No yaw response to the motors ramping up and down (t=[167,170]s)
2. Altitude loss 
3. Thrust saturation 
4. No more yaw control

[image]
[FlightReview link](http://logs.px4.io/plot_app?log=28ee3ee8-5a91-448e-853e-bcf058a70986)

bresch, 2017-05-29:
I maybe found the issue in the `mixer_multirotor.cpp` file ([link](https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp#L326-L337)).
```
} else if (out > 1.0f) {
			// allow to reduce thrust to get some yaw response
			float prop_reduction = fminf(0.15f, out - 1.0f);
			// keep the maximum requested reduction
			thrust_reduction = fmaxf(thrust_reduction, prop_reduction);

			if (fabsf(_rotors[i].yaw_scale) <= FLT_EPSILON) {
				yaw = 0.0f;

			} else {
				yaw = (1.0f - ((roll * _rotors[i].roll_scale + pitch * _rotors[i].pitch_scale) *
					       roll_pitch_scale + thrust + boost)) / _rotors[i].yaw_scale;
}
```
When the output saturates, the `thrust_reduction` is computed but the new yaw value is then computed again with the old unreduced thrust. The final result is then a reduced thrust with no yaw control.

bresch, 2017-05-29:
I fixed the computation of the new yaw value with the modified thrust and flew three times like a crazy pilot to reach the saturations and now it's fine. I'll do a PR for the fix.
You can see the logs [here](http://logs.px4.io/plot_app?log=08e124ec-c9d0-4205-a740-9e48006290d9), [here](http://logs.px4.io/plot_app?log=4105252f-3d0b-48a7-9190-93a3e0cc8769) and [here](http://logs.px4.io/plot_app?log=c3a2cb92-66c6-40fd-8a95-8c560f8ac490)

closed 2017-05-30 by bresch

Stifael, 2017-05-30:
@bresch: 
what I still don't get why the velocity setpoint is at zero for both vx and vy despite the fact that the vehicle is in position control using just velocity control loop:
[image]

Stifael, 2017-05-30:
and once you switch the mode, everything is fine. seems like there is another issue independent of the mixture...

bresch, 2017-05-30:
@Stifael Could you explain a bit more the issue you are pointing out please? I don't get it. And from which log is your plot?

Stifael, 2017-05-30:
@bresch: it is from the log that you posted. 
why did you assume from the beginning that just the yaw control was lost?

bresch, 2017-05-30:
@Stifael You are right. I found that it coincides with the value of the x-y RC stick going to the central position. Do you think it is related?
[image]
(The RC value not shown is at 0)

mhkabir, 2017-05-30:
If the sticks are at zero, shouldn't the vel sp still have some value because it's trying to hold position against external disturbances?

Stifael, 2017-05-30:
not necessarily. if the vehicle has not entered position control, then the velocity setpoint will be zero. The vehicle enters position control only if the stick are close to zero and the velocity is low

Stifael, 2017-05-30:
so the param that defines the threshold for entering position control is MPC_HOLD_MAX_XY, which is set to 0.8. But the velocity in y is ~3m/s, so the vehicle should not have entered position control and therefore is completely fine to have a velocity setpoint of 0. Nonetheless, still strange to have such a bad tracking right it setpoint goes to zero

bresch, 2017-05-30:
The position tracing was really bad in this log because I had a misalignment of the mag heading and the frame heading. The correction was then projected into a slightly rotated frame which causes a "circling around position" error. The issue isn't on the control side.