PX4/PX4-Autopilot #7289: FW_LND_USETER

hamishwillee, 2017-05-25:
[FW_LND_USETER](https://dev.px4.io/en/advanced/parameter_reference.html#FW_LND_USETER) description says:

> Use terrain estimate during landing

Can we add a comment to this parameter:

> If set (and valid) the terrain altitude from GPS is used to determine the ground altitude during landing.

tubeme, 2017-05-25:
I think it is the Baro Altitude used for the landing and not the GPS Alt. I don't know if we use GPOS.Alt for the landing...

hamishwillee, 2017-05-25:
@tubeme - either way, the current docs are clearly not correct/sufficient. 

This information was from @dagar - can you confirm what is correct?

hamishwillee, 2017-05-30:
Thanks @dagar

see3peoh, 2017-07-05:
Is this affected by having a lidar on board? Terrain estimator is disabled by default now, is it simply best to not use it? If so, how does the lidar assist in terrain estimation landing?

santiago3dr, 2018-08-01:
this issue was resolved prior to 1.8.0 release
here's the pr that addressed it
https://github.com/PX4/Firmware/pull/8412

closed 2018-08-01 by santiago3dr