PX4/PX4-Autopilot #7291: the problem of tecs.cpp

garlinplus, 2017-05-25:
The function as followed:

void TECS::_update_speed_demand()
{
	// Set the airspeed demand to the minimum value if an underspeed condition exists
	// or a bad descent condition exists
	// This will minimise the rate of descent resulting from an engine failure,
	// enable the maximum climb rate to be achieved and prevent continued full power descent
	// into the ground due to an unachievable airspeed value
	if ((_badDescent) || (_underspeed)) {
		_TAS_dem     = _TASmin;
	}

	// Constrain speed demand
	_TAS_dem = constrain(_TAS_dem, _TASmin, _TASmax);

	// calculate velocity rate limits based on physical performance limits
	// provision to use a different rate limit if bad descent or underspeed condition exists
	// Use 50% of maximum energy rate to allow margin for total energy controller
	float velRateMax;
	float velRateMin;

	if ((_badDescent) || (_underspeed)) {
		velRateMax = 0.5f * _STEdot_max / _integ5_state;
		velRateMin = 0.5f * _STEdot_min / _integ5_state;

	} else {
		velRateMax = 0.5f * _STEdot_max / _integ5_state;
		velRateMin = 0.5f * _STEdot_min / _integ5_state;
	}

//	// Apply rate limit
//	if ((_TAS_dem - _TAS_dem_adj) > (velRateMax * _DT)) {
//		_TAS_dem_adj = _TAS_dem_adj + velRateMax * _DT;
//		_TAS_rate_dem = velRateMax;
//
//	} else if ((_TAS_dem - _TAS_dem_adj) < (velRateMin * _DT)) {
//		_TAS_dem_adj = _TAS_dem_adj + velRateMin * _DT;
//		_TAS_rate_dem = velRateMin;
//
//	} else {
//		_TAS_dem_adj = _TAS_dem;
//
//
//	_TAS_rate_dem = (_TAS_dem - _TAS_dem_last) / _DT;
//	}

	_TAS_dem_adj = constrain(_TAS_dem, _TASmin, _TASmax);

	//xxx: using a p loop for now
	_TAS_rate_dem = constrain((_TAS_dem_adj - _integ5_state) * _speedrate_p, velRateMin, velRateMax);

//	_TAS_dem_last = _TAS_dem;

}

the bug as followed:

	if ((_badDescent) || (_underspeed)) {
		velRateMax = 0.5f * _STEdot_max / _integ5_state;
		velRateMin = 0.5f * _STEdot_min / _integ5_state;

	} else {
		velRateMax = 0.5f * _STEdot_max / _integ5_state;
		velRateMin = 0.5f * _STEdot_min / _integ5_state;
	}

dagar, 2017-06-27:
Bad descent isn't implemented. What do you think the velocity rate limits should be for underspeed mode?

garlinplus, 2017-09-21:
The code is not wrote by me.It's always as this.
I mean no mater what the value (_badDescent) || (_underspeed) is,all the code  
	velRateMax = 0.5f * _STEdot_max / _integ5_state;
	velRateMin = 0.5f * _STEdot_min / _integ5_state; 
will be excuted.So why we need to use if function?

dagar, 2017-10-08:
It was likely intended to have different limits, but never implemented.

closed 2018-02-21 by PX4BuildBot