PX4/PX4-Autopilot #7292: the problem of TECS::_detect_underspeed() in tecs.cpp

garlinplus, 2017-05-25:
the function as followed:
void TECS::_detect_underspeed()
{
	if (!_detect_underspeed_enabled) {
		_underspeed = false;
		return;
	}

	if (((_integ5_state < _TASmin * 0.9f) && (_throttle_dem >= _THRmaxf * 0.95f)) || ((_integ3_state < _hgt_dem_adj)
			&& _underspeed)) {

		_underspeed = true;

	} else {
		_underspeed = false;
	}
}

the bug :+1: 
if (((_integ5_state < _TASmin * 0.9f) && (_throttle_dem >= _THRmaxf * 0.95f)) || ((_integ3_state < _hgt_dem_adj)&& _underspeed)) {

		_underspeed = true;

	} else {
		_underspeed = false;
	}

_integ3_state < _hgt_dem_adj&& _underspeed 

when _integ3_state <_hgt_dem_adj and _underspeed ==ture

the code will set _underspeed = ture

dagar, 2017-06-27:
Why is that a bug? 

If you're underspeed and throttle is max set underspeed.
OR
If your height is below the demand and you're already underspeed, continue setting it.

garlinplus, 2017-09-21:
when _underspeed  is true,it's still set true.

dagar, 2017-10-08:
Yes, that's odd, it makes the integ3 condition useless.