PX4/PX4-Autopilot #7293: the problem in the TECS::_update_energies()

garlinplus, 2017-05-25:
The function code as followed:

void TECS::_update_energies(void)
{
	// Calculate specific energy demands
	_SPE_dem = _hgt_dem_adj * CONSTANTS_ONE_G;
	_SKE_dem = 0.5f * _TAS_dem_adj * _TAS_dem_adj;

	// Calculate specific energy rate demands
	_SPEdot_dem = _hgt_rate_dem * CONSTANTS_ONE_G;
	_SKEdot_dem = _integ5_state * _TAS_rate_dem;

	// Calculate specific energy
	_SPE_est = _integ3_state * CONSTANTS_ONE_G;
	_SKE_est = 0.5f * _integ5_state * _integ5_state;

	// Calculate specific energy rate
	_SPEdot = _integ2_state * CONSTANTS_ONE_G;
	_SKEdot = _integ5_state * _vel_dot;
}

In above code the variable named  _SPEdot_dem denotes the speed of change of the Gravity potential energy, And the variable name _SKEdot_dem means the speed of change of kinetic energy.
With Kinetic energy theorem Ek = 1/2*m*v^2 ，
we can know _SKE_dem = 0.5f * _TAS_dem_adj * _TAS_dem_adj.
So the speed of change of kinetic energy is the derivative of _SKE_dem.
_SKEdot_dem  = 0.5*2*_SKE_dem*_(_SKE_dem)' =SKE_dem*_(_SKE_dem)' 
Because _integ5_state  is estimate of true airspeed,we should not use _integ5_state to represent the  the derivative of _SKE_dem.But we can use _integ4_state  to replace it ,because it's derivative of _integ5_state.

   We can verify it by the formula of _SKEdot.As the function says _SKEdot = _integ5_state * _vel_dot.
What is _vel_dot?We can find it in the follow function.

void TECS::update_state(float baro_altitude, float airspeed, const math::Matrix<3,3> &rotMat,
	const math::Vector<3> &accel_body, const math::Vector<3> &accel_earth, bool altitude_lock, bool in_air)
{
	// Implement third order complementary filter for height and height rate
	// estimted height rate = _integ2_state
	// estimated height     = _integ3_state
	// Reference Paper :
	// Optimising the Gains of the Baro-Inertial Vertical Channel
	// Widnall W.S, Sinha P.K,
	// AIAA Journal of Guidance and Control, 78-1307R

	// Calculate time in seconds since last update
	uint64_t now = ecl_absolute_time();
	float DT = max((now - _update_50hz_last_usec), UINT64_C(0)) * 1.0e-6f;

	// printf("dt: %10.6f baro alt: %6.2f eas: %6.2f R(0,0): %6.2f, R(1,1): %6.2f\naccel body: %6.2f %6.2f %6.2f\naccel earth: %6.2f %6.2f %6.2f\n",
	// 	DT, baro_altitude, airspeed, rotMat(0, 0), rotMat(1, 1), accel_body(0), accel_body(1), accel_body(2),
	// 	accel_earth(0), accel_earth(1), accel_earth(2));

	bool reset_altitude = false;

	if (_update_50hz_last_usec == 0 || DT > DT_MAX) {
		DT = DT_DEFAULT; // when first starting TECS, use small time constant
		reset_altitude = true;
	}

	if (!altitude_lock || !in_air) {
		reset_altitude = true;
	}

	if (reset_altitude) {
		_integ3_state = baro_altitude;
		_integ2_state = 0.0f;
		_integ1_state = 0.0f;

		// Reset the filter state as we just switched from non-altitude control
		// to altitude control mode
		_states_initalized = false;
	}

	_update_50hz_last_usec = now;
	_EAS = airspeed;

	_in_air = in_air;

	// Get height acceleration
	float hgt_ddot_mea = -(accel_earth(2) + CONSTANTS_ONE_G);
	// Perform filter calculation using backwards Euler integration
	// Coefficients selected to place all three filter poles at omega
	float omega2 = _hgtCompFiltOmega * _hgtCompFiltOmega;
	float hgt_err = baro_altitude - _integ3_state;
	float integ1_input = hgt_err * omega2 * _hgtCompFiltOmega;
	_integ1_state = _integ1_state + integ1_input * DT;
	float integ2_input = _integ1_state + hgt_ddot_mea + hgt_err * omega2 * 3.0f;
	_integ2_state = _integ2_state + integ2_input * DT;
	float integ3_input = _integ2_state + hgt_err * _hgtCompFiltOmega * 3.0f;

	// If more than 1 second has elapsed since last update then reset the integrator state
	// to the measured height
	if (reset_altitude) {
		_integ3_state = baro_altitude;

	} else {
		_integ3_state = _integ3_state + integ3_input * DT;
	}

	// Update and average speed rate of change
	// Only required if airspeed is being measured and controlled
	float temp = 0;

	if (PX4_ISFINITE(airspeed) && airspeed_sensor_enabled()) {
		// Get DCM
		// Calculate speed rate of change
		// XXX check
		temp = rotMat(2, 0) * CONSTANTS_ONE_G + accel_body(0);
		// take 5 point moving average
		//_vel_dot = _vdot_filter.apply(temp);
		// XXX resolve this properly
		_vel_dot = 0.95f * _vel_dot + 0.05f * temp;

	} else {
		_vel_dot = 0.0f;
	}

	if (!_in_air) {
		_states_initalized = false;
	}

}

In the above function
	
if (PX4_ISFINITE(airspeed) && airspeed_sensor_enabled()) {
		// Get DCM
		// Calculate speed rate of change
		// XXX check
		temp = rotMat(2, 0) * CONSTANTS_ONE_G + accel_body(0);
		// take 5 point moving average
		//_vel_dot = _vdot_filter.apply(temp);
		// XXX resolve this properly
		_vel_dot = 0.95f * _vel_dot + 0.05f * temp;

	} else {
		_vel_dot = 0.0f;
	}

So we can know _vel_dot  is the accelerator of x axis.So we can confirm that the formula "_SKEdot_dem = _integ5_state * _TAS_rate_dem;" is error. Because the _integ5_state  is velocity,not a accelerator .

garlinplus, 2017-06-03:
@JamesHarrison @lfelipe @NaterGator @lucasdemarchi @hauptmech

dagar, 2017-06-27:
I believe that's correct.

    K = 1/2 * v^2 (per unit mass)
    dK/dt = v * dv/dt

garlinplus, 2017-09-21:
Yes,your formula is correct.
But my opinion is that the formula _SKEdot_dem = _integ5_state * _TAS_rate_dem is wrong
not  _SKEdot = _integ5_state * _vel_dot.

closed 2018-02-21 by PX4BuildBot