PX4/PX4-Autopilot #7298: SITL Gazebo unable to echo Gazebo Topics

ganeshrk95, 2017-05-25:
I have the latest master clones. When I run the IRIS SITL using:

make posix_sitl_default gazebo

After the model loads and PX4 shell opens, I did:

gz topic -l 

I get the following
[image]

I just cannot echo any iris related topics. It returns nothing. I'm developing a Tiltrotor Tricopter model for SITL. I need to view what's going into the motor_speed topics to debug my code.

closed 2018-02-21 by PX4BuildBot

mchen0037, 2018-06-28:
Hi, did you ever find out more about this? Thanks

TSC21, 2018-06-28:
@mchen0037 https://github.com/PX4/sitl_gazebo/issues/102