PX4/PX4-Autopilot #7303: tilt-rotor switch throttle problem

lfynevermore, 2017-05-27:
We found that when the tilt-rotor switched from mc mode to the fixed wing mode, the throttle would drop to its lowest point，it caused the speed of motors drop down, and then go back to normal about 0.1 seconds later. Is this caused by switching logic? Will it affect the flight?
[image]

garlinplus, 2017-05-27:
it  should be switched in fix wing mode

lfynevermore, 2017-05-28:
yes,it switched from mc mode to fw mode, but i want to know why it occurred a 0.1 seconds' throttle drop at the last of switch transition, how can i avoid this process, or increase the throttle immediately at the moment when it switched to fw mode? is it related to the flight mode in mc mode?

garlinplus, 2017-05-28:
because in the mc mide,the input of motor(in fact its esc)include pitch  control quantity and  throttle quantity.But when it's switched  in fix wing mode,the pitch control  in only  decided by  elevator ,so the control  output  decrese.

lfynevermore, 2017-05-28:
it can be shown in the figure, the throttle decreased to the lowest point(900 PWM), it is not only caused by pitch control, i think it seems like a restart process, and maybe caused by the switch logic.

garlinplus, 2017-05-28:
are you chinese?

lfynevermore, 2017-05-28:
yes 发自网易邮箱大师 在2017年05月28日 11:30，garlinplus 写道： are you chinese?

closed 2018-02-21 by PX4BuildBot