PX4/PX4-Autopilot #7305: Empty log after crash - v1.6.0 RC3

Shpiler, 2017-05-27:
When does it write a log file to sd card? After crash I found log file empty. The only useful thing I got is that txt:

2017_05_27_08_48_14: main state transition denied
2017_05_27_08_50_15: ARMED by arm/disarm component command
2017_05_27_08_50_25: Takeoff detected
2017_05_27_08_50_30: [logger] file: /fs/microsd/log/2017-05-27/08_50_1
2017_05_27_08_50_40: Mag #0 fail:  TOUT!

but given it ended up upside-down with GPS/Mag module smashed in pieces... "Mag #0 fail" looks obvious. (Mag0 is the external one, right?) 

Is there any chance to recover log? I can't believe it keeps the entire log in RAM until disarm.

bkueng, 2017-05-29:
The log is continuously written to the SD card (with a very short buffer). If you don't find the log, it most likely means the file system on the SD card got corrupted. You might be able to get it with some recovery tool.

LorenzMeier, 2017-06-04:
Sorry to hear about your crash. Let us know if you can recover the log. I'm closing this for now since there is nothing we can do without that log file.

closed 2017-06-04 by LorenzMeier