PX4/PX4-Autopilot #7309: Stuck during mission, if altitude higher than max height

ChristophTobler, 2017-05-29:
For a survey I did, the default altitude was higher than the max height and the quad got stuck because it never reached the waypoint.
Here's the [log](http://logs.px4.io/plot_app?log=c828d9ab-c6de-4034-a4ce-9ea5222faf74)

sanderux, 2017-06-11:
@Stifael this remains a problem in missions. Could you please fix this or change your implementation to be more like geofence?

Stifael, 2017-06-11:
@sanderux, I dont really want to take the credit for the altitude limit, because I dont like to have it in the landdetector either and would like to remove it. However, I dont know how geofence works right now and not sure if I have the time to work on that. What I can do right now is to limit the mission  altitude to the max altitude...

Stifael, 2017-06-11:
@LorenzMeier I am in favor of turning off the landdetector alt max for now, since in order to implement it properly it requires some more changes. Currently the landdetector adjusts max altitude based on battery level. Lets assume the vehicle is flying a mission. It is possible to adjust the desired altitude to the max altitude, but what happens if the vehicle is in between two waypoints and the battery level goes below the threshold which results in a lower max altitude. The navigator  already has set the current waypoint and does not know about the altitude max change and will therefore set a mission altitude above the max.
In order to solve that, the navigator also needs to know when the max altitude has changed and then update the mission item accordingly. However, before I implement something like that I prefer to clean up the navigator first such that waypoints are actually only sent when they have changed.

Stifael, 2017-06-11:
Actually, the navigator knows when something in the landdetector had changed. So I guess I can do it without much change for mission. But the same check has to be done for all the other auto related maneuvers such as rtl etc.,

sanderux, 2017-06-12:
@Stifael based on https://github.com/PX4/Firmware/commit/a1982e039288221c7ecac62e4ea3ebf909b2578e i was under the impression that it was you who added this to the land detector. In a later commit you enabled it by default.

I agree it would be best to disable it for now. I never understood what it's doing in the land detector anyway.

Stifael, 2017-06-12:
@sanderux that is correct, but I just followed orders. I never wanted to put it there either

Stifael, 2017-06-16:
here is a branch where I moved the altitude limit to the navigator and removed the altitude limit from the position controller and land-detector.
https://github.com/PX4/Firmware/tree/pr-limit_altitude_navigator

the reason why I did that:
- a detector should just detect an event, but should not apply any action as it was done in the land-detector for altitude limitation
- the battery level is already handled/detected by the battery.cpp class. Hence, it is not necessary to do another detection in the land-detector
- I removed the altitude limit from the position controller;
I still don't really see/understand why the position controller needs to make any adjustments and should rather track the desired set-points.  I do not really see the benefit of limiting altitude, velocity (as it is done when being close to the home_pose)  in manual control mode, since the user should be in full control when using the sticks. In fact, with the current implementation the user cannot even fly below home_pose with full speed since it is limited. 

it is to note that the altitude limitation in the navigator only applies for mission related maneuvers such as mission, rtl etc. It does not work yet for repositioning such as altitude change and go-to, since they are unfortunately handle differently. I think we also should change that such that any navigator related maneuver follows the same procedure (=different topic).

closed 2017-10-10 by Stifael