PX4/PX4-Autopilot #7311: MS5611 Navio2 sudden temperature and/or pressure jumps

nicolaerosia, 2017-05-29:
Hello,

I'm experiencing sudden jumps in temperature and/or pressure jumps on RPIv3 + Navio2.
It looks like that from time to time, the read value for temperature is 0, in which case the next value to be read will be invalid (pressure). The same applies for pressure reading, if it is zero, the next reading (temperature), will be invalid.
Since a sample is composed of a temperature & pressure reading, in the worst case, when pressure is zero, we will miss two samples.

I've addressed this here https://github.com/PX4/DriverFramework/pull/198

mswingtra, 2017-07-18:
We have observed very similar issue on pixracer. This occurs on average once every ~30-50 bootups.
Currently, we're based off stable v1.5.5, but I believe it is also here on master since data collection logic did not really change since then.
In our case, pressure reading [here](https://github.com/PX4/Firmware/blob/master/src/drivers/ms5611/ms5611.cpp#L829) becomes negative, and then [power function some lines below](https://github.com/PX4/Firmware/blob/master/src/drivers/ms5611/ms5611.cpp#L875) returns NaN because the base is negative.

We're going to address it in a very similar way by skipping publishing round if measurement is not finite, but I guess it would be interesting to dig in further and determine why pressure reading becomes negative once in a while.

dagar, 2017-07-18:
Looking at the driver for the first time I notice it's nearly the same as the ms5525 differential pressure sensor I recently added. In that case any adc read (raw) returning 0 is invalid. https://github.com/PX4/Firmware/blob/master/src/drivers/ms5525_airspeed/MS5525.cpp#L197

I'm travelling this week, but I can take a closer look at this when I return.

dagar, 2017-07-18:
[image]

http://www.te.com/commerce/DocumentDelivery/DDEController?Action=srchrtrv&DocNm=MS5611-01BA03&DocType=Data+Sheet&DocLang=English

nicolaerosia, 2017-07-18:
I've fixed this on Navio2 by dropping Work Queues and reworking the measurement logic: https://github.com/PX4/DriverFramework/pull/215

CarlOlsson, 2017-09-08:
Hi!
@dagar Did you manage to find the cause of this issue on Pixracer? @nicolaerosia whats the status regarding your PR for Navio2?

nicolaerosia, 2017-09-08:
@CarlOlsson check here the commits for MS5611: https://github.com/nicolaerosia/DriverFramework/commits/mpu9250-rework
Currently I don't have the bandwidth to push that into DF and honestly it's not worth it. @dagar is trying to get the NuttX drivers working on Linux, but those are not great either. (atleast ms5611).
Take a look at my branch, the jitter I'm seeing is under 30 cm.
If you want less jitter, remove the sleep from this function:
https://github.com/nicolaerosia/DriverFramework/blob/3dce84255f6c450e593e13ea57e6921ae605cb3a/drivers/ms5611/MS5611.cpp#L595

and you'll have only 
```
while(1) {
instance->_measure();
}
```

CarlOlsson, 2017-09-08:
Okey, will look into it and see if it fixes the issue, thanks!

nicolaerosia, 2017-09-08:
let me know if it fixes your issue

CarlOlsson, 2017-09-08:
Absolutely!

dagar, 2017-09-08:
This PR will make the ms5611 cross platform - https://github.com/PX4/Firmware/pull/7907
Then I'd like to go through out and implement @nicolaerosia's best practices.

http://www.freesteel.co.uk/wpblog/2014/11/26/ms5611-altitude-sensor-very-sensitive-to-own-working-temperature/

nicolaerosia, 2017-10-25:
@CarlOlsson any updates?

CarlOlsson, 2017-10-29:
Sorry didn't see your comment.
Due to time constraints we had to solve the issue by writing a simple outlier detector before we feed the data to the EKF. Will update here if/when we get the time to go for a cleaner fix

closed 2018-02-21 by PX4BuildBot