PX4/PX4-Autopilot #7312: 1.6.0 Problem: Loiter Change Altitude

tubeme, 2017-05-29:
In PX4 1.6.0 In FW if the plane is in Loiter and we set change altitude in QGC, once the change is confirmed with the slider the plane changes its current center dot to the point where the slider conformation is done so the loiter continues with a new center spot.

My expectation is once the change altitude is confirmed in QGC to see the plane gain or loose altitude with the current circle  that it is doing in loiter at the moment.

dagar, 2017-05-30:
This is arguably more of a QGC issue as it's sending a new REPOSITION (with lat, lon) each time. 
We could discuss implementing MAV_CMD_DO_CHANGE_ALTITUDE, but it's missing on both sides.

tubeme, 2017-05-30:
I Understand. In this situation PX4 is not very useful regarding professional observation work. With a proffessional observation work we have a lot of situations where we can spend hours in loiter in order to have a better look at something. A lot of times it involves gaining or loosing altitude. It does not matter if we use Zoom camera or not. Zoom has its limitations and also the stability of the gimbal could prevent high zoom levels in our form factor. So the only solution is to rise or lower the altitude without loosing the center spot.

dagar, 2017-05-30:
@DonLakeFlyer any opinion for how you'd like to handle this situation?
Either implement MAV_CMD_DO_CHANGE_ALTITUDE or allow altitude only changes in MAV_CMD_DO_REPOSITION?
I can implement the PX4 side.

LorenzMeier, 2017-06-04:
Not a QGC issue:
```
    vehicle->sendMavCommand(vehicle->defaultComponentId(),
                            MAV_CMD_DO_REPOSITION,
                            true,   // show error is fails
                            -1.0f,
                            MAV_DO_REPOSITION_FLAGS_CHANGE_MODE,
                            0.0f,
                            NAN,
                            NAN,
                            NAN,
                            vehicle->homePosition().altitude() + newAltRel);
```

Fix needs to happen on the PX4 side @dagar

LorenzMeier, 2017-06-04:
Fix is here: https://github.com/PX4/Firmware/pull/7351

I'm closing this.

closed 2017-06-04 by LorenzMeier

BazookaJoe1900, 2017-07-10:
@LorenzMeier, I can see that #7351 fixes this issue, but its not in this milestone, is it going to be?