PX4/PX4-Autopilot #7319: land_detector detects takeoff/landing with motors disconnected

nicolaerosia, 2017-05-30:
Hello,

As I'm preparing for my first flight, I'm trying to check as many things as I can and I've stumbled upon this issue.
Worth mentioning is that propellers are not attached.

If I do commander arm, raise the throttle, then land_detector detects takeoff, and landing when I lower the throttle.

Log here: http://logs.px4.io/plot_app?log=3381ef1b-73e3-4cc7-afa6-a24028f7dabc
[16_34_54.zip](https://github.com/PX4/Firmware/files/1039092/16_34_54.zip)

Any ideas?

dagar, 2017-05-30:
Manual input is a factor for manual modes.
https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L207

mhkabir, 2017-05-30:
This is not really a bug. The system assumes that it's taking off because it has no other feedback to make a decision with. GPS isn't accurate enough to make that decision either.

nicolaerosia, 2017-05-31:
Can't we use the accelerometer somehow?

bresch, 2017-05-31:
Having a false positive in the takeoff detector is not a big problem. However, a detection with an accelerometer (in addition to the manual input) might lead to some false negatives which are much more unwanted. A proper tuning of the takeoff thrust value is then a good conservative solution.

LorenzMeier, 2017-06-02:
This all works as designed.

closed 2017-06-02 by LorenzMeier