PX4/PX4-Autopilot #7322: odom.twist.linear is in body frame

marcelinomalmeidan, 2017-05-30:
I am not sure if this is an issue, but I couldn't find a reason for this to be.
The message that I get in Mavros for local odometry has linear position expressed in (local) world frame, while linear velocity is expressed in body frame. Is there a reason for this to be?
It would be more intuitive to me if linear velocity was expressed in some inertial frame, instead of body frame.

mhkabir, 2017-05-30:
As far as I remember, it's a ROS standard. PX4 sends the data in the inertial frame. You should check the mavros transforms. 

Please reopen on mavros side if desired.

closed 2017-05-30 by mhkabir

marcelinomalmeidan, 2017-05-30:
@mhkabir, you are right!
According with odom documentation, this is how it is supposed to be:

> _This represents an estimate of a position and velocity in free space.  
> The pose in this message should be specified in the coordinate frame given by header.frame_id.
> The twist in this message should be specified in the coordinate frame given by the child_frame_id_
> Header header
> string child_frame_id
> geometry_msgs/PoseWithCovariance pose
> geometry_msgs/TwistWithCovariance twist

Thanks for pointing that out!