PX4/PX4-Autopilot #7324: Unclear firmware version <=> hardware correspondance in documentation

anuppari, 2017-05-30:
It is unclear which firmware version should be used with the [Pixhawk Mini](https://store.3dr.com/products/3dr-pixhawk) (i.e., FMUv2, FMUv3, or FMUv4).
The [documentation page](https://docs.px4.io/en/flight_controller/pixhawk_series.html) lists it as v4, however QGC refuses to flash v4 citing incompatible board ID:
[image]

This [release page](https://github.com/PX4/Firmware/releases/tag/v1.5.1) suggests it should be v3, though it lists it as "Pixhawk (2) Mini", even though 3DR nor the documentation pages list it as being a "2", and the pixhawk 2 is a completely different device.

When connecting a new pixhawk mini to QGC, it lists it as having FMUv2. Even after flashing v3 (i.e., "custom" option in QGC), QGC still reads it as having v2 onboard. QGC allows the mini to be flashed by v2 and v3, so it's not clear which one to use:
[image]

So which version should be used with the pixhawk mini?

anuppari, 2017-05-30:
Also, QGC seems to automatically download v2, so is that the correct one?

[image]

mhkabir, 2017-05-31:
v2 or v3 for the Pixhawk Mini. The v3 target uses the full 2MB flash. 

I will fix the documentation in a minute. Sorry about the confusion.

anuppari, 2017-05-31:
What is the full flash used for? I've seen some mention of needing a different firmware for using the LPE instead of the EKF, does v3 include both?

mhkabir, 2017-06-01:
Yes v3 includes both

dagar, 2017-12-06:
I believe this has been improved. Please reopen if I'm wrong.

charles-blouin, 2017-12-21:
Hi,

I would like to confirm that the V3 seems like the right firmware. The V2 outputs an error (GRYO 0 not found). V4 refuses to install. The documentation [here](https://docs.px4.io/en/flight_controller/pixhawk_series.html#fmu-versions) on the wiki seems wrong.

dagar, 2017-12-21:
@charles-blouin which board do you have and which version of the Firmware did you try?

charles-blouin, 2017-12-21:
This is with a Pixhawk mini. I used the px4fmu-v3_default.px4 firmware on the [github](https://github.com/PX4/Firmware/releases) release page, which Qgroundcontrol reports as 1.7.0. I was able to perform the calibrations and correctly display the attitude in Qgroundcontrol.

I am new to the PX4 environment. When I received the Pixhawk Mini, I flashed the new firmware using the default firmware in Qground control, and I kept obtaining the error Preflight fail: no gyro sensor 0. I spend a few hours researching this. The information of the wiki I linked above confused me. If I connect the Pixhawk Mini in Qgroundcontrol, the board reports as a PX4 FMU V2 in the firmware tab. Then, when I flash the firmware using the default PX4 flight stack, the gyro is not recognized and the calibration fails. I tried to manually flash the firmware for the FMU v4 using the Custom Firmware file option, but that report a board id error and the firmware does not upload. In the end, only the firmware for the v3 FMU worked.

LorenzMeier, 2017-12-21:
@santiago3dr Could you re-test with QGC and Pixhawk Mini?

charles-blouin, 2017-12-21:
Additionnal information:
Here is a screenshot of what happens when I plugin the Pixhawk mini:
[image]
When I was flashing the firmware using the default configuration as in the image above, Qgroundcontrol was downloading the v2 firmware version, and it had the Preflight fail: no gyro sensor 0.
Now that I have flashed the v3 version, when I reflash the board, it defaults to the V3 firmware, which works.

santiago3dr, 2017-12-21:
There was an issue with the pixhawk mini running 1.7.1 that flagged the error of No gyro and No accel found.
Said error was fixed in 1.7.2 as per commit b718901

Successfully flashed stable/default firmware on pixhawk mini with qgroundcontrol:
[image]
[image]

charles-blouin, 2017-12-21:
I would like to update the [gitbook](https://docs.px4.io/en/flight_controller/pixhawk_series.html#fmu-versions) documentation. Can I confirm that the correct firmware is the px4fmu-v3_default.px4? Is there a reason Qgroundcontrol specifies FMU v2 and the gitbook says it is FMU v4?

dagar, 2017-12-22:
@charles-blouin px4fmu-v2 and px4fmu-v3 are nearly the same, except the v3 has access to the full 2MB of flash. PX4 includes more modules (like UAVCAN) only in the v3. If you look in the firmware you'll see that px4fmu-v3_default is actually px4fmu-v2_large.

Due to the bootloader that the pixhawk mini ships with we can't automatically flash fmu-v3 through QGC, but you can still do it manually. 

I've opened https://github.com/PX4/px4_user_guide/pull/121 to fix the documentation. Sorry about the confusion.

closed 2017-12-22 by dagar

LorenzMeier, 2017-12-22:
The Gitbook docs seem correct, because it correctly states that FMUv4 is Pixracer.

hamishwillee, 2017-12-22:
@LorenzMeier FYI, that text just got updated following this issue.

KJBB, 2018-11-02:
After floundering for months, I see a post from someone who is likely to be able to help. I have posted on various forums without success.

I have two Pixracer flight controllers, and neither has any SERIAL_ or FRSKY_ parameters present in the parameter list. I've just reflashed the "latest" via QGC - http//px4-travis.s3.amazonaws.com/firmware/stable/px4fmu-v4_default.px4.
No luck. A search for SER or FRSKY in the parameter list draws a blank.

I tried to use GitHub and forked and cloned multi-megabytes. I tried to find the makefile I wanted, but ended up guessing which was the correct one. I ran it anyway and it started and then said something was missing! Damn.
I feel I have been running several marathons in sequence, just to get firmware that I had expected to be there in the first place. I am exhausted, confused, and contemplating buying a new Pixracer just to see what I get!
I think I "updated" the firmware on each drone on first use, and possibly threw away the parameters I want? I'm extremely pi.... off, and lost as to my next step.
I noted the comment "Is there a reason Qgroundcontrol specifies FMU v2" which makes me wonder if QGC is limiting my choice somehow, despite the px4fmu-v4_default.px4 result that appears to be the best a man can get?

Do you think you can rescue a lost soul?
Ken Barrie - Scotland.