PX4/PX4-Autopilot #7325: Internal mag data on pixracer board is not rotated correctly

priseborough, 2017-05-31:
**SW versions affected:**

Detected on 1.6.0 stable.

**HW affected:**

Detected on pixracer (px4fmu-v4)

**Description:**

When the SENS_BOARD_ROT  = 4 (board pointing backwards with ESC connectors to the front) and the sensors are calibrated, the X,Y and Z magnetometer  data appears to be swapped. This results in large yaw errors during flight and loss of position control requiring reversion to manual or altitude hold.

bresch, 2017-05-31:
I'm running px4fmu_v4pro and I also have heading issues. My autopilot is rotated by 90 degrees and the external mag by 180 degrees. After calibration, I always have an offset with the true North.
@mhkabir already found several issues related to the mag calibration and orientation during the last few days, but it seems that it's not completely fixed.

priseborough, 2017-06-01:
After further investigation, the SENS_EXT_MAG parameter has to be set to 2 if no external magnetometer is connected, otherwise the sensor module applies the wrong rotation to the data from the sensor_mag topic.

The SENS_EXT_MAG = 0 option is not working correctly.

mhkabir, 2017-06-01:
@priseborough, are you testing master? I fixed that issue on master a few days back and removed that parameter. It shouldn't even be present on master at the moment. 

Please retest.

mhkabir, 2017-06-01:
OK sorry - no. I was talking about a different parameter. I can check this particular issue out and fix it in case you already haven't by the time I reach home.

My airline somehow lost the bag with all my autopilot stuff so it might take some time for me to come back up to speed :P

priseborough, 2017-06-01:
After further checking on a another board (using master), with SENS_EXT_MAG = 2 and SENS_BOARD_ROT = 4, I have confirmed that the data reported on the sensor_mag topic is rotated correctly into the autopilot frame of reference, but that the mag data reported on sensor_combined has been rotated incorrectly into the body reference frame.

priseborough, 2017-06-01:
BTW, I could not access the SENS_EXT_MAG parameter using QGC and had to modify it using the nsh console. The sensor_mag and sensor_combined topics were monitored using the nsh prompt listener utility.

priseborough, 2017-06-01:
I have confirmed that explicitly setting CAL_MAG0_ROT = 0 instead of the default value of -1 fixes the error.

priseborough, 2017-06-01:
The mag rotation matrix from autopilot to body frame is being calculated in three places. 

When parameter updates are checked. This should set the mag rotation matrix to be equal to the board rotation matrix for an internal sensor.

https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L405
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L454

Where the mag sensor is polled. 

https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L747

This is incorrect because it will use whatever rotation is returned by a   -1 index. This code should be moved to this section so that the rotation matrix is not calculated unless the parameter is >=0

https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L466

priseborough, 2017-06-01:
Fix submitted here: https://github.com/PX4/Firmware/pull/7331

PR #7337 (2017-06-02): Correctly handle internal magnetometer rotations
@priseborough @LorenzMeier This properly fixes #7325. Please test.
closed 2017-06-02 by LorenzMeier