PX4/PX4-Autopilot #7326: CMD_OVERRIDE_GOTO does not work

imrhch, 2017-05-31:
It seems to me that current implementation of CMD_OVERRIDE_GOTO does not work.
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L957-L964

When I send the command in order to pause processing of the mission, the commander writes
WARN  [commander] Pause mission cmd
and positive acknowledge is returned, but the vehicle continues.
I suppose that changing of status_local->nav_state does not change the control state.

As many other commands use main_state_transition function, is it possible that this handling of CMD_OVERRIDE_GOTO is obsolete?

dagar, 2017-05-31:
Yes that's pretty bogus and should be deleted. Try MAV_CMD_DO_REPOSITION instead.

imrhch, 2017-06-01:
Thanks! :)
MAV_CMD_DO_REPOSITION with all values NAN pauses the mission and restart is possible with MAV_CMD_MISSION_START, I just need to keep actual mission item index.

PR #7356 (2017-06-09): remove partial CMD_OVERRIDE_GOTO implementation
 - closes #7326
commit 8d5c955a (2017-06-09): remove partial CMD_OVERRIDE_GOTO implementation (#7356)

- closes #7326
closed 2017-06-09 by dagar