PX4/PX4-Autopilot #7327: IMU/Magnetometer does not start on battery power

anuppari, 2017-05-31:
Using the pixhawk mini, the imu and magnetometer seem to not start up consistently. Sometimes (using the wireless telemetry or USB with QGC) I can see the roll/pitch and heading indicators don't update. Also, IMU messages are not published by mavros. Stabilized mode also doesn't work.
It's usually fixed by just disconnecting and reconnecting the battery or USB cable a few times.

This is on Release 1.6 and using the latest (may 29) daily build of QGC, but I've been having this same problem for a few weeks with 1.6 RC3.

anuppari, 2017-05-31:
I've also had this issue on multiple pixhawk mini's, so I don't think it's a hardware issue.

anuppari, 2017-05-31:
I've tried uploading a log but I keep getting error 400 failed to parse errors

dagar, 2017-05-31:
Are you using the new logger?
Set `SYS_LOGGER 1` and `SDLOG_MODE 1` to log immediately at boot.

anuppari, 2017-05-31:
They were both already set to using the new logger and log from boot to shutdown. Is there anything I should be doing before pulling power to make sure the log file doesn't get corrupted if it is continuously logging until shutdown?

LorenzMeier, 2017-06-04:
@anuppari If you arm and disarm the disarming will stop the log and prevent it getting corrupted.

Also make sure to use a QGC daily build - otherwise the file ending of the log file will be incorrect.

LorenzMeier, 2017-06-04:
Also, I believe I fixed this yesterday - please re-test with 1.6.1

anuppari, 2017-06-06:
I tested this with 1.6.1 and still have the same issue with the IMU and mag not starting up.

anuppari, 2017-06-06:
Can we reopen this?

dagar, 2017-06-09:
Yes, but we'll need more information.

anuppari, 2017-06-27:
[Here's a log](http://review.px4.io/plot_app?log=75786cc9-9ecd-42ae-bb1d-1fd686069510) showing the issue. Note the roll/pitch estimates do not show anything, and the raw mag output is zero.

What additional info do you need?

dagar, 2017-06-27:
Can you try connecting the console via QGroundControl? Then you can try manually starting and stopping each driver. https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L94

We need to figure out why these drivers are failing to start sometimes, but unfortunately I don't see a lot to go on from the existing logs.

santiago3dr, 2017-06-27:
ran bench test on a pixhawk mini with 1.6.3-stable
http://logs.px4.io/plot_app?log=3d0a07f5-14e9-4ae3-be42-c243f5df3744

no gps/external mag connected
accel,gyro and internal mag (MPU9250) looks to be working just fine

anuppari, 2017-06-27:
We get the issue intermittently, so usually after a power cycle, it works fine. We are also using FMUv3 with the LPE estimator, in case that's relevant, though it seems to be a sensor issue.

Were there any relevant changes from 1.6.1 to 1.6.3? It doesn't seem like a one-off hardware issue as we have the exact same problem on 5 separate pixmini's

santiago3dr, 2017-06-27:
re-tested with fmu-v3 and lpe estimator; same results... will keep testing
http://logs.px4.io/plot_app?log=0d4787c1-57b3-4e4a-9074-2142fd02594a

gbrunso, 2017-06-27:
anuppari and I tried resetting, starting, and stopping the sensor driver (mpu9250) from the mavlink console in QGC.
If the sensors start up correctly, the sensors and the console commands work appropriately.  If the sensors do not start up, trying to reset or turn the drivers off and on does not seem to have any effect.  
I updated to 1.6.3 and we are still having the same problem.

dagar, 2017-06-27:
@anuppari which binary do you flash? I just noticed you're using LPE, which isn't in the default fmu-v2 binary.

First try changing SYS_MC_EST_GROUP to 2. 

Can you try the latest flashing with the latest QGC daily build that only recently added support for flashing fmu-v3s?

anuppari, 2017-06-27:
We are flashing v3, which based on #7324, can be used on the pixhawk mini to enable the LPE.

anuppari, 2017-06-27:
And yes we've been using a QGC daily build from a few weeks ago

dagar, 2017-06-27:
Have you been manually flashing it?

In your log I see accel and gryo, but no mag. The attitude indicator should still be updating, unless the estimator isn't starting. 
Are you using an external mag? Is it always connected?

anuppari, 2017-06-27:
Yes we are using the custom firmware option in QGC to flash the firmware.

The attitude and heading indicator both don't update in QGC when this issue happens. We are not using an external mag, just the pixhawk mini, without GPS/compass.

Is there a way to check if the attitude estimator is starting up correctly? From the log it appears the mag sensor is not starting up, so I think the issue is upstream of the estimator.

anuppari, 2017-06-27:
A similar issue seems to have [occurred previously](https://github.com/PX4/DriverFramework/commit/2eb3d9bdd11a01bf26f40a35c42c3f8104a4f4e4). Not sure if that fix is related.

dagar, 2017-06-27:
Ok, so it sounds like the mpu9250 is periodically failing to start for you, but the attitude indicator is a mystery.

Can you use the mavlink console in QGC when it happens to see what's running? It's in the log download tab.
Run ps, and ls /dev. Try it with and without the attitude indicator updating.

With access to the system console this would be easier because we could just see what's happening during boot.

anuppari, 2017-06-27:
Working on that now.

Regarding the attitude indicator, isn't the IMU and mag all contained in one mavlink message? Does the attitude estimator wait until it receives at least one mag uOrb message before sending out an attitude estimate mavlink message?

Is there a way to access the system console during boot?

gbrunso, 2017-06-27:
Here is the console output for ps and ls /dev when the issue is happening.
`
NuttShell (NSH)
nsh> ps
  PID PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK  COMMAND
    0   0 FIFO     Kthread --- Ready              00000000 Idle Task
    1 192 FIFO     Kthread --- Waiting  Signal    00000000 hpwork
    2  50 FIFO     Kthread --- Waiting  Signal    00000000 lpwork
    3 100 FIFO     Task    --- Waiting  Semaphore 00000000 init
  292 250 FIFO     Task    --- Waiting  Semaphore 00000000 lp_estimator
  103 220 FIFO     Task    --- Waiting  Semaphore 00000000 gps 1
  296 250 FIFO     Task    --- Waiting  Semaphore 00000000 mc_att_control
  329 175 FIFO     pthread --- Waiting  Semaphore 00000000 mavlink_rcv_if2 0x200117a0
  298 250 FIFO     Task    --- Waiting  Semaphore 00000000 mc_pos_control
  107  90 FIFO     Task    --- Waiting  Semaphore 00000000 dataman
  172 100 FIFO     Task    --- Waiting  Signal    00000000 mavlink_if0 mavlink start -r 1200
  173 175 FIFO     pthread --- Waiting  Semaphore 00000000 mavlink_rcv_if0 0x100072e0
  334  50 FIFO     pthread --- Waiting  Semaphore 00000000 commander_low_prio 0x0
  335 100 FIFO     Task    --- Running            00000000 mavlink_shell
  240 250 FIFO     Task    --- Waiting  Semaphore 00000000 logger -b 16 -t -f logger start
  241  60 FIFO     pthread --- Waiting  Semaphore 00000000 log_writer_file 0x20010520
  306 105 FIFO     Task    --- Waiting  Semaphore 00000000 navigator
  149 250 FIFO     Task    --- Waiting  Semaphore 00000000 sensors
  151 140 FIFO     Task    --- Waiting  Signal    00000000 commander commander start
  184 100 FIFO     Task    --- Waiting  Signal    00000000 mavlink_if1 mavlink start -d /dev/ttyS2 -b 57600 -m osd -r 1000
  185 175 FIFO     pthread --- Waiting  Semaphore 00000000 mavlink_rcv_if1 0x2000dbb0
  284 250 FIFO             --- Waiting  Semaphore 00000000 attitude_estimator_q
  158 240 FIFO     Task    --- Waiting  Semaphore 00000000 px4io
  223 100 FIFO     Task    --- Waiting  Signal    00000000 mavlink_if2 mavlink start -r 800000 -d /dev/ttyACM0 -m config -x
nsh> ls /dev
/dev:
 accel0
 adc0
 baro0
 console
 gyro0
 led0
 mag0
 mmcsd0
 mpu9250_accel
 mpu9250_gyro
 mpu9250_mag
 ms5611_spi_int
 mtdblock0
 mtdblock1
 null
 pipe0
 pipe1
 pwm_output0
 pwm_output1
 px4fmu
 px4io
 ram0
 rgbled0
 tone_alarm0
 ttyACM0
 ttyS0
 ttyS1
 ttyS2
 ttyS3
 ttyS4
 ttyS5
 ttyS6
nsh> `

gbrunso, 2017-06-27:
This should be a little easier to read.

NuttShell (NSH)
nsh> ps
  PID PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK  COMMAND
    0   0 FIFO     Kthread --- Ready              00000000 Idle Task
    1 192 FIFO     Kthread --- Waiting  Signal    00000000 hpwork
    2  50 FIFO     Kthread --- Waiting  Signal    00000000 lpwork
    3 100 FIFO     Task    --- Waiting  Semaphore 00000000 init
  292 250 FIFO     Task    --- Waiting  Semaphore 00000000 lp_estimator
  103 220 FIFO     Task    --- Waiting  Semaphore 00000000 gps 1
  296 250 FIFO     Task    --- Waiting  Semaphore 00000000 mc_att_control
  329 175 FIFO     pthread --- Waiting  Semaphore 00000000 mavlink_rcv_if2 0x200117a0
  298 250 FIFO     Task    --- Waiting  Semaphore 00000000 mc_pos_control
  107  90 FIFO     Task    --- Waiting  Semaphore 00000000 dataman
  172 100 FIFO     Task    --- Waiting  Signal    00000000 mavlink_if0 mavlink start -r 1200
  173 175 FIFO     pthread --- Waiting  Semaphore 00000000 mavlink_rcv_if0 0x100072e0
  334  50 FIFO     pthread --- Waiting  Semaphore 00000000 commander_low_prio 0x0
  335 100 FIFO     Task    --- Running            00000000 mavlink_shell
  240 250 FIFO     Task    --- Waiting  Semaphore 00000000 logger -b 16 -t -f logger start
  241  60 FIFO     pthread --- Waiting  Semaphore 00000000 log_writer_file 0x20010520
  306 105 FIFO     Task    --- Waiting  Semaphore 00000000 navigator
  149 250 FIFO     Task    --- Waiting  Semaphore 00000000 sensors
  151 140 FIFO     Task    --- Waiting  Signal    00000000 commander commander start
  184 100 FIFO     Task    --- Waiting  Signal    00000000 mavlink_if1 mavlink start -d /dev/ttyS2 -b 57600 -m osd -r 1000
  185 175 FIFO     pthread --- Waiting  Semaphore 00000000 mavlink_rcv_if1 0x2000dbb0
  284 250 FIFO             --- Waiting  Semaphore 00000000 attitude_estimator_q
  158 240 FIFO     Task    --- Waiting  Semaphore 00000000 px4io
  223 100 FIFO     Task    --- Waiting  Signal    00000000 mavlink_if2 mavlink start -r 800000 -d /dev/ttyACM0 -m config -x
nsh> ls /dev
/dev:
 accel0
 adc0
 baro0
 console
 gyro0
 led0
 mag0
 mmcsd0
 mpu9250_accel
 mpu9250_gyro
 mpu9250_mag
 ms5611_spi_int
 mtdblock0
 mtdblock1
 null
 pipe0
 pipe1
 pwm_output0
 pwm_output1
 px4fmu
 px4io
 ram0
 rgbled0
 tone_alarm0
 ttyACM0
 ttyS0
 ttyS1
 ttyS2
 ttyS3
 ttyS4
 ttyS5
 ttyS6
nsh>

dagar, 2017-06-27:
Interesting, I only see the mpu9250, but the pixhawk mini is supposed to also have an ICM20608.

Can you do the same thing for when it's not working?

dagar, 2017-06-27:
@santiago3dr do you see both the mpu9250 and ICM20608 running on your pixhawk mini?

gbrunso, 2017-06-27:
The output of ls /dev is the same when everything starts up correctly.

anuppari, 2017-06-27:
Those outputs were for when the mag was not working. The outputs are the same when it is working.

Starting the ICM20608 seems to be just an [additional parameter sent to mpu9250](https://github.com/PX4/Firmware/blob/a6ef6c6e33edd63977b990cf8f973a255ebe8ac5/ROMFS/px4fmu_common/init.d/rc.sensors#L110). Should that launch a separate process?

anuppari, 2017-06-27:
Nevermind, I guess the same process can make additional devices available.

In any case, the ICM20608 doesn't have a magnetometer.

Also, the dev folder lists accel0, gyro0, and mag0. What are those?

dagar, 2017-06-28:
I'm at a bit of a loss here, and don't have one of these to debug with.

For sanity can you completely wipe all params (QGC reset all), flash latest PX4 master, and try again. Please try both LPE and EKF2. 

To summarize what little we know so far.
1. mpu9250 consistently starting, but sometimes doesn't publish mag
2. lpe + attq seem to be always running, but occasionally aren't publishing anything
3. ICM20608 doesn't seem to be starting

Other things to try.
 - `mpu9250 stop`, `mpu9250 start`, `mpu9250 test`
 - `mpu6000 -T 20608 start`, `mpu6000 -T 20608 stop`, `mpu6000 -T 20608 test`
 - `sensors test`

santiago3dr, 2017-06-28:
testing mpu on console, getting error
ERROR [mpu9250] immediate mag read failed

this error is appearing on some tests but not all; very inconsistent results

mhkabir, 2017-06-28:
Have you considered that the hardware may be bad? Can you test with a second board if you have one?

anuppari, 2017-06-28:
I've had this identical issue on 5 separate pixhawk mini's, so I don't think it's a hardware issue unless 3DR recently made some board revision and we got a bad batch. Where would board revisions be listed, I don't see it under [hardware](https://github.com/PX4/Hardware)

dagar, 2017-06-28:
@anuppari have you tried resetting all your params and flashing the latest PX4 master? Have you tried using EKF2 instead of LPE or a different airframe?

There must be something specific about your setup if it's consistent across 5 pixhawk minis, but so far @santiago3dr hasn't been able to reproduce. We need to isolate that piece to drill down into the root cause. Right now there's just not enough information available to debug.

gbrunso, 2017-06-28:
In the mavlink console, I tried running mpu9250 start and mpu9250 test when the pixhawk mini booted up correctly and incorrectly.

When the it started up correctly:

nsh> mpu9250 start
MPU9250 on SPI bus 1 at 4 (1000 KHz)
WARN  [mpu9250] probe failed! 255
WARN  [mpu9250] no device on bus 4

nsh> mpu9250 test
WARN  [mpu9250] single read
WARN  [mpu9250] time:     42441530
WARN  [mpu9250] acc  x:  	  0.1277	m/s^2
WARN  [mpu9250] acc  y:  	  1.5872	m/s^2
WARN  [mpu9250] acc  z:  	 -9.5418	m/s^2
WARN  [mpu9250] acc  x:  	56	raw 0x38
WARN  [mpu9250] acc  y:  	331	raw 0x14b
WARN  [mpu9250] acc  z:  	-1957	raw 0xf85b
WARN  [mpu9250] acc range: 156.9064 m/s^2 ( 16.0000 g)
WARN  [mpu9250] gyro x: 	 -0.00330	rad/s
WARN  [mpu9250] gyro y: 	  0.00448	rad/s
WARN  [mpu9250] gyro z: 	  0.00688	rad/s
WARN  [mpu9250] gyro x: 	14	raw
WARN  [mpu9250] gyro y: 	22	raw
WARN  [mpu9250] gyro z: 	15	raw
WARN  [mpu9250] gyro range:  34.9066 rad/s (2000 deg/s)
WARN  [mpu9250] temp:  	 58.1163	deg celsius
WARN  [mpu9250] temp:  	8345	raw 0x2099
WARN  [mpu9250] mag x: 	  0.24427	rad/s
WARN  [mpu9250] mag y: 	 -0.02293	rad/s
WARN  [mpu9250] mag z: 	 -0.00419	rad/s
WARN  [mpu9250] mag x: 	139	raw
WARN  [mpu9250] mag y: 	-155	raw
WARN  [mpu9250] mag z: 	-25	raw
WARN  [mpu9250] mag range:  48.0000 Ga
WARN  [mpu9250] mag temp:   58.0997	deg celsius

When the it did not start up correctly:

nsh> mpu9250 start
MPU9250 on SPI bus 1 at 4 (1000 KHz)
AK8963: bad id
AK8963: failed to read adjustment data
WARN  [mpu9250] probe failed! 255
WARN  [mpu9250] no device on bus 4

nsh> mpu9250 test
WARN  [mpu9250] ret: -1, expected: 72
ERROR [mpu9250] immediate acc read failed
ERROR [mpu9250] Task exited with errno=0

santiago3dr, 2017-06-28:
trying with a second board, I'm getting the same results
mpu9250 usually starting properly but on some tests, mag values not available
when mpu fails to start in console, getting: ERROR [mpu9250] immediate acc read failed

also tried latest ardupilot beta that supports the pixhawk mini and both primary accel/gyro's look to be responsive when plotting; mpu9250 mag also showing values

-both master and stable of 1.6 do log values for said sensors when armed for duration of log

dagar, 2017-06-28:
The mpu9250 immediate read failure might be unrelated. Cheap the output of listener.

    listener sensor_accel 0 0
    listener sensor_accel 0 1

    listener sensor_gyro 0 0
    listener sensor_gyro 0 1

gbrunso, 2017-06-29:
We just flashed the latest px4 v3 master and reset all our configurations and can still replicate the issue using both lpe and ekf.

dagar, 2017-06-29:
Thanks for trying that @gbrunso 
@davids5 would you mind taking a look at this with your pixhawk mini? It's difficult to debug remotely without one.

gbrunso, 2017-06-30:
We just tried 1.6.3 v2 default and were also able to replicate the error.

davids5, 2017-06-30:
@dagar I will try to get this on my bench today. If not early next week

gbrunso, 2017-07-03:
We noticed that our sensor start-up failure tends to happen more frequently when we power the pixhawk with the battery, and then connect the usb.  Since it can be powered with both usb and the power distribution board, could it cause problems to have them both connected?

davids5, 2017-07-03:
@gbrunso Thank you for the additional information. I would think they can both be connected without a problem.  I will have a look at it today.

LorenzMeier, 2017-07-08:
@gbrunso This should be fixed in the referenced PR: https://github.com/PX4/Firmware/pull/7554 - would you mind trying?

gbrunso, 2017-07-12:
I just flashed the latest px4 v3 master and can still replicate this issue. Unlike before, when the sensors fail to start up, QGC shows the following messages:
```

Warning: Primary compass not found
PREFLIGHT FAIL: NO ACCEL SENSOR 0
Warning: Primary accelerometer not found
PREFLIGHT FAIL: NO GYRO SENSOR 0
Warning: Primary gyro not found
```

davids5, 2017-07-12:
I think there is some confusion caused by several bugs/changes. So let me list what I know to expect.

1)  The pxfmu-v3_default build prior to  https://github.com/PX4/Firmware/commit/22d0ea15deb00d7ead6db54fe491571e3e6a4035 did not start the ICM20608 on the PixhawkMini. Current master does start it. N.B. pxfmu-v2_default will not start it.

2) Prior to PR: #7554 when the MPU9250 internal mag failed the the driver was falsely listed in `/dev`
 after #7554 - when the MPU9250 internal mag fails the complete devier does not start. The thinking is something is wrong with the chip or system state. We need to not mask the error.

3) @r0gelion indicated if the external GPS is connected. That it is easy to be fooled via QGC feedback.  The new GPS has a mag in it, and since the cabling is not as it use to be with the older 3DR GPS/MAG. The older units (FMUv2) has 2 cables (serial and I2C )the newer only has 1 with both serial and I2C.  So it is possible someone might not realize a mag is connected or for that matter easily disconnect just the mag. 

So please begin tests with the GPS disconnected!

Other notable points and things to check:

1) @r0gelion has indicated he has one board that fails 100% and one that works.

My board works 100% of the time. 

@gbrunso noted:
>We noticed that our sensor start-up failure tends to happen more frequently when we power the pixhawk with the battery, and then connect the usb. Since it can be powered with both usb and the power distribution board, could it cause problems to have them both connected?

@r0gelion would you be able to see if the order of powering Battery then USB is a factor as noted by on both units you have.

@anuppari noted:
> I've had this identical issue on 5 separate pixhawk mini's, so I don't think it's a hardware issue unless 3DR recently made some board revision and we got a bad batch. Where would board revisions be listed, I don't see it under hardware

@anuppari did you try master after #7554 went in? Can you test on the 5 units?

if the Firmware was just loaded and you have a board that fails - please do a complete (nothing connected that can back feed either) power cycle. 

Thank you for helping track this down.

anuppari, 2017-07-13:
@gbrunso and I are working with the same 5 physical units, so yes we have tested the latest master and still get the issue. Also, we are testing indoors without the GPS module connected.

davids5, 2017-07-13:
@anuppari 

I am sorry I did not realise you were working with @gbrunso. Thank you both for helping troubleshoot this problem. I very much appreciate your help.  

Do I understand you correctly that all 5 boards fail? Is this 100% of the time or randomly? Can you describe the details of the power sequencing and what is connected to board in what order that leads to the failure? Also can you include a picture of the power module you are using?

gbrunso, 2017-07-14:
They fail randomly, but there seems to be a greater tendency to fail if the board has been off for a length of time, and then powered with the battery.  We can usually get the sensors to start if we disconnect the battery to power off, power with USB, and then reconnect the battery.  We are using the PDB that comes with the pixhawk mini.  It looks like this:
[image]

davids5, 2017-07-15:
@gbrunso - THANK YOU!

Your observation helped me find the root cause and the reason the results of testing were so inconsistent.  Please test https://github.com/PX4/Firmware/pull/7618

davids5, 2017-07-15:
arrg, I am still getting failures. I am sorry please hold off on testing.

anuppari, 2017-07-17:
We are using these boards indoors with mocap integration via setting `LPE_FUSION`, `ATT_EXT_HDG_M` and `ATT_W_EXT_HDG` appropriately. Is there a way to circumvent this issue in the short term by having the attitude and local position estimator ignore the the mag/compass all together and just work even if the compass doesn't start up? This would be really useful in the short term.

davids5, 2017-07-21:
I have found the issue it is a hardware problem on the pixhawkMini that makes the mpu9250 unreliable. 

There is no FW workaround. I spent quite a bit trying to create a 100% reliable workaround, that would  resolve the issue, but in the end, the mpu9250 is still unreliable. 

Since we now have the ICM20608 running and the kit comes with an external mag, we have deprecated support the mpu9250 on the pixhawkMini HW.  This is done in [PR](https://github.com/PX4/Firmware/pull/7618)  See [this](https://github.com/PX4/Firmware/pull/7618/commits/e242775613cb478a7f395555772f1c2268af5c1d) commit.

gbrunso, 2017-07-27:
Thank you @davids5, so does the latest master for the pixhawk mini just not use the mpu9250?

mhkabir, 2017-07-27:
@gbrunso I think it's not in master yet : https://github.com/PX4/Firmware/pull/7618.

davids5, 2017-07-27:
@gbrunso as @mhkabir noted it is still out on a branch on PR #7618 for @PX4TestFlights to check.

gbrunso, 2017-08-04:
I flashed PR #7618 and still have the intermittent error.  When it happens, QGroundControl shows the error "Critical Warning: Primary compass not found" and the attitude indicators in QGC stay at zero.

davids5, 2017-08-04:
@gbrunso 

With that PR, using a v3 build (the only one that should be used on a mini) you will need to use the external GPS that has the mag on it. 

Is it connected?

gbrunso, 2017-08-04:
No, we are not using the external GPS.

[18 more replies]

closed 2018-02-21 by PX4BuildBot