PX4/PX4-Autopilot #7333: UAVCAN restart fails on v1.6.0

tstellanova, 2017-06-01:
- Running build from git tag v1.6.0:  `make px4fmu-v3_default upload`
- Running on Pixhawk 2.1 hardware
- Connected to nsh on CONS port 
- `param set UAVCAN_ENABLE 2` and `param save`
- Connect Zubax GNSS 2 to Pixhawk 2.1 via CAN, reboot
- From nsh, command: `uavcan stop` followed by `uavcan start`
- uavcan reports endless errors on nsh console (see below).
- Reboot and uavcan will start again

```
WARN  [uorb] node_open as advertiser failed.
uavcan_mag ADVERTISE FAILED
uavcan_mag adding channel 124...
WARN  [uorb] node_open as advertiser failed.
uavcan_mag ADVERTISE FAILED
uavcan_mag adding channel 124...
WARN  [uorb] node_open as advertiser failed.
uavcan_mag ADVERTISE FAILED
```

mhkabir, 2017-06-01:
@bryanmonty anything on this?

mhkabir, 2017-06-01:
@bryanmonti @pavel-kirienko Do you know anything about this? `Advertise failed` usually points to something leaking file descriptors.

pavel-kirienko, 2017-06-01:
I will have a look into this someday. This is not a critical issue because UAVCAN should never be stopped in a production setting.

bryanmonti, 2017-06-01:
I was able to confirm the issue. I ran `uavcan stop` then `uavcan start fw`, returned the same issue as above.

LorenzMeier, 2017-06-02:
Definitely a resource leak, reproducible on a Pixhawk 1 by stopping / starting 2-3 times. I did poke a bit around but I didn't find it on the first pass.

closed 2018-02-21 by PX4BuildBot