PX4/PX4-Autopilot #7334: UAVCAN GNSS2 not detected on Pixhawk 2.1 CAN1 port

tstellanova, 2017-06-01:
- Connect Zubax GNSS2 to Pixhawk 2.1 CAN1 port
- Install px4fmu_v3 (`make px4fmu-v3_default upload` using git tag v1.6.0)
- Reboot Pixhawk 2.1
- Using nsh, `uavcan info` shows:
```
nsh> uavcan info
Pool allocator status:
	Capacity hard/soft: 500/250 blocks
	Reserved:  3 blocks
	Allocated: 3 blocks
UAVCAN node status:
	Internal failures: 0
	Transfer errors:   0
	RX transfers:      0
	TX transfers:      691
CAN1 status:
	HW errors: 55619
	IO errors: 55619
	RX frames: 0
	TX frames: 461
CAN2 status:
	HW errors: 43
	IO errors: 43
	RX frames: 287
	TX frames: 461
ESC actuators control groups: sub: 0 / req: 0 / fds: 2
ESC mixer: NONE
Sensor 'gnss':
RX errors: 0, using old Fix: 1, receiver node id: N/A

Sensor 'mag':
devname: /dev/mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'baro':
devname: /dev/baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Online nodes (Node ID, Health, Mode):
```
- Disconnect Zubax GNSS2 and reconnect to CAN2 port on Pixhawk 2.1
- Reboot
- After a few seconds,`uavcan info` shows a valid node ID for the GNSS :
```
Sensor 'gnss':
RX errors: 0, using old Fix: 1, receiver node id: 124
```

Possibly related to this outstanding Ardupilot issue: https://github.com/ArduPilot/ardupilot/issues/5272
The common factor seems to be the Pixhawk 2.1 hardware and any UAVCAN driver code shared between the two projects.

pavel-kirienko, 2017-06-01:
The dynamic node ID allocation server is available only on CAN1. Pixhawk 2.1 has the labels CAN1 and CAN2 swapped, so you're actually supposed to use the interface marked as CAN2. The issue is hardware-related and well known, so I'm closing this.

mhkabir, 2017-06-02:
This isn't well known. @pavel-kirienko can you please open a pull request to PX4 Devguide with this information?

pavel-kirienko, 2017-06-07:
@mhkabir I would, but the documentation for Pixhawk 2.1 seems to be published at http://www.proficnc.com/ and not at the dev guide. @proficnc is already aware of this issue. Please advise what to do.

closed 2018-02-21 by PX4BuildBot