PX4/PX4-Autopilot #7336: logger: cant set rate for topics

CarlOlsson, 2017-06-02:
Hi,

If I put the file logger_topics.txt in etc/logging/ on an SD card with the following lines everything works as expected
```
sensor_gyro
sensor_accel
```

However, if I try to set the the interval like below, then the log file is always 0 bytes.
```
sensor_gyro, 500
sensor_accel, 500
```
I'm using a Pixhawk pro and latest master

mhkabir, 2017-06-02:
@CarlOlsson try removing the space between the comma and the number. ;)

CarlOlsson, 2017-06-02:
The spaces should be there according to the [doc](https://dev.px4.io/en/log/logging.html)
However, also tried to remove them but then the logger is logging the default topics and not the ones that are defined in the file

mhkabir, 2017-06-02:
OK, yes you're correct. The spaces are fine.

This issue seems to be a real issue. Let me have a look.

dagar, 2017-06-02:
Isn't it with a space like this?

    vehicle_attitude 50
    vehicle_attitude_setpoint 50

LorenzMeier, 2017-06-02:
This is most likely a regression of a NuttX update - `sscanf("%s %u")` might not be matching any more.

bkueng, 2017-06-02:
I just tested on Pixracer, it works fine. The comma is optional.
There was a NuttX update regression, but I already fixed that several weeks ago.

LorenzMeier, 2017-06-02:
Assuming @CarlOlsson is using a stale version, closing.

closed 2017-06-02 by LorenzMeier

CarlOlsson, 2017-06-06:
@bkueng strange that it works for you and not for me. I just tested it again. 
1) Flashed master via QGC to a Pixhawk Pro, 
2) Put this file [logger_topics.txt](https://github.com/PX4/Firmware/files/1054859/logger_topics.txt) in etc/logging on the SD card
3) Plug in the SD card in the Pixhawk Pro, plug in a usb cable to power it. 
4) Check SD card -> Logfile is zero bytes.

But if you instead put the file without the interval time it works as expected
[logger_topics.txt](https://github.com/PX4/Firmware/files/1054874/logger_topics.txt)

I also tested the same thing on a Pixracer with the same results

CarlOlsson, 2017-06-06:
Here is the logfile from the test that works, i.e. no setting of time interval
http://logs.px4.io/plot_app?log=752d73f0-37b9-4dc1-96c0-bb24db09a310

bkueng, 2017-06-06:
Really strange. Your topics file works for me & the log looks as expected.
What's the boot output? What does `logger stop`, `logger start` say?