PX4/PX4-Autopilot #7338: Inspect flight log

LorenzMeier, 2017-06-02:
@priseborough Can you inspect this log which has issues with the position estimate?

http://review.px4.io/plot_app?log=98d340b4-d820-4819-af6d-48acd5f9ff77

priseborough, 2017-06-03:
There were severe problems with the magnetometer that caused the copter to lose it's yaw reference when hovering. Here is a plot of the field strength:

[image]

Here are the mag field and mag yaw innovations:

[image]

[image]

priseborough, 2017-06-03:
I might be able to improve resistance to this sort of error by not switching back into heading fusion if the magnetometer is failing innovation consistency checks, but with mag data this bad it is only delaying the inevitable until the combination of mag error and vehicle orientation fools the checks.

closed 2018-02-21 by PX4BuildBot