PX4/PX4-Autopilot #7352: GPIO Camera kills AUX PWM

tops4u, 2017-06-04:
This is a Duplicate to #7103

I have configured GPIO Triggering on Pin 5 to low while I have a seconday Connection to a Optocoupler on +5V which triggers my FLIR Cam. My AUAV-X2 does only have 5 AUX Pins so I had to slightly modify the default GPIO Setting for the Camera Trigger (instead of using GPIOs 5-6). In #7103 it was mentioned that AUX PWM **AND** GPIO should work if the default Pins 5/6 are left untouched. 

With this Setup again with 1.6 Release Build all other AUX Pins will not output any PWM Signal. 

If this is a non Valid Configuration please implement one of the following Logic Schemes:
- Remove RC to AUX Settings as they are no longer valid
- Display a Warning / Info that this can not be used in conjunction with AUX Passthru from RC
- Fix this in a way that Pin 5 can be used as a single GPIO Pin while AUX 1-4 work with PWM

mhkabir, 2017-06-05:
If you had left the setting at default (56), it would've worked. Even though pin 6 is not exposed on the AUAV-X2, it doesn't make any functional difference to the autopilot code. 

This is noted in BIG **bold** letters on the camera trigger page : https://dev.px4.io/en/tutorials/camera_trigger.html, but it seems like I need to add a warning which prevents people from shooting themselves in the foot.

tops4u, 2017-06-05:
Ok Thanks for the Answer. Sorry but some things are commented in Code -> Visible in QGC, there is a User Guide and a Dev Guide. 
To be honest as User I don't Expect that I'll have to lookup too many things in the Dev Guide, so I don't always consult all documentation. 
Apart from this I fully understand that there are technical reasons for these limitations - but I don't see why using 5&6 works while only 5 does not since this configuration of mine would not violate the restriction. Since there was no Pin 6, it did not sound logic to me to use the default.

I'll try and test this later today.

tops4u, 2017-06-05:
@mhkabir That works, thanks.

closed 2017-06-05 by tops4u