PX4/PX4-Autopilot #7358: pixracer mag offset

urlu75, 2017-06-05:
I was using PX4 1.5.5 without any major issue. Upgrade to 1.6.x and now have an 180 degrees offset with the mag after recalibrating all the sensors.
I don't know if it's related, but I can't go down when altitude hold is enable and position control is flying away (probably the mag issue).

Downgrade to 1.5.5 with the same setting, recalibrate all sensors and everything is fine again.

mhkabir, 2017-06-05:
Are you running 1.6.1 which we released yesterday? We fixed an issue with the mag rotation there. 

So please describe your setup, external mag rotation, etc.

urlu75, 2017-06-05:
I tried the 1.6.1 this afternoon.
I do not have the hexa with me to dump the setting (sorry for the low detail).
The board is front/back inverted (yaw180) and external compass (HMC5983) has no rotation (YAW 0). X4R SBUS receiver, frsky telemetry on frsky port, mavlink telemetry for OSD on port 1 and normal telemetry @19k on port 2. Almost all the parameters were by default with this testing.

mhkabir, 2017-06-05:
@urlu75 Did you reboot after calibration? Please provide a parameter dump too.

matanhavi, 2017-06-06:
@mhkabir note that there still a bug.
if you change parameter (any parameter, for example I changed the SDLOG_UTC_OFFSET) before flight, for some reason the SENS_BOARD_ROT 'loses' its value and the rotation is wrong.
you can easily test it on table, put SENS_BOARD_ROT as non default value (i had 2), reset,
look at the heading, calibrate if need.
change SDLOG_UTC_OFFSET, and look at the heading.

mhkabir, 2017-06-06:
@matanhavi OK that parameter not saving thing sounds concerning. Let me have a look.

mhkabir, 2017-06-06:
I just checked both your issues - 

> if you change parameter (any parameter, for example I changed the SDLOG_UTC_OFFSET) before flight, for some reason the SENS_BOARD_ROT 'loses' its value and the rotation goes wrong.
you can easily test it on table, put SENS_BOARD_ROT as non default value (i had 2), reset,
look at the heading, calibrate if need.
change SDLOG_UTC_OFFSET, and look at the heading.

@matanhavi I could not reproduce this. What hardware are you running? From what vendor?

> Upgrade to 1.6.x and now have an 180 degrees offset with the mag after recalibrating all the sensors.

@urlu75 I just flashed a fresh board and replicated your sensor setup. Everything looks fine on my side and I wasn't able to reproduce it - I extensively tested a bunch of rotations and corner cases. Please reflash your board using QGC with 1.6.1 to make sure you're on the latest version.

matanhavi, 2017-06-06:
@mhkabir I am using regular Pixhawk 2.1. and version 1.6.1
Try reproduce by this:
- use the Pixhawk on table, no other HW needed (also not external compass)
- set SENS_BOARD_ROT to YAW 90
- calibrate (if needed)
- reset
- look at highres_IMU.xmag,ymag,zmag
- change SDLOG_UTC_OFFSET to anything
- look at highres_IMU.xmag,ymag,zmag, ymag goes to xmag value, xmag goes to -xmag
[image]
- if you reset again, all goes back to "normal"

mhkabir, 2017-06-06:
Still couldn't reproduce. I tried with a Pixhawk 2.1 : 

[image]

matanhavi, 2017-06-06:
I done some more test, from fresh parameters of Pixhawk i couldn't reproduce it too.
but after I set my parameter file, it does reproduced. I am trying to find why

matanhavi, 2017-06-06:
I handpick some parameters from my AP with that issue.
and reproduce it with minimal changes from default parameters of new AP. (airframe not even set... and calibration is not real, i toke it from other AP)
I attached both files, on that is just default parameters after set to default from QGC
the other is one that i changed and reproduce the issue
[P_badparams.txt](https://github.com/PX4/Firmware/files/1054674/P_badparams.txt)
[default_parmas.txt](https://github.com/PX4/Firmware/files/1054675/default_parmas.txt)

matanhavi, 2017-06-06:
@mhkabir I have more info.
its seems that at first when the AP load it uses SENS_BOARD_ROT as the internal magnetometer rotation.
but after some change of parameter, VotedSensorsUpdate::parameters_update() is called, but there, the CAL_MAG0_ROT is used instead.

i think it might be related to on of your last commits, 
https://github.com/PX4/Firmware/commit/6ee09ca16fd5e52237b4566f6d10de525cc2cb9e

mhkabir, 2017-06-06:
The thing is - after a change of parameters, you are supposed to reboot. QGC doesn't force this yet, but we will add it soon. Can you see if you can reproduce the issue if you reboot the autopilot after any changes to a the CAL_MAG* or SENS_BOARD_ROT parameter(s)?

mhkabir, 2017-06-06:
@mhkabir OK I think I found a bug. Stay tuned for testing the fix.

matanhavi, 2017-06-06:
"The thing is - after a change of parameters, you are supposed to reboot."  - That is not true for few reasons:
1) there are some internal logics that changes parameters, for example, the flight time, BAT_N_CELLS, and other (there is even param_set_no_notification() on this code)
2) how do you supposed to calibrate PID, for example with this restriction
3) the reason for each 'parameters_update()' on modules is to allow parameter change on the fly, those logics must be safe.

I agree that for some parameters its advised not to change "on air" and you sure must to reboot. having mechanism to lock them can be nice.

I am not sure now about my previous comment.

mhkabir, 2017-06-06:
What I meant was, after changing the calibration and orientation parameters - https://github.com/PX4/Firmware/commit/49890c61f5c98be9aaf5371fb5d29a8168b2a0d6

Please do as I say and try rebooting after setting orientations and/or calibrating and see if you can still reproduce the bad mag rotation problem.

matanhavi, 2017-06-06:
I am rebooting after calibrate. calibrating goes to "I agree that for some parameters its advised not to change "on air" and you sure must to reboot" statement.
on all my tests I didn't calibrate at all.
Can you go to gitter now?

mhkabir, 2017-06-06:
Sure. Let's chat.

mhkabir, 2017-06-06:
OK I found the issue. Some old code in `parameters_update()` was resetting the rotation matrices unnecessarily. I fixed it here : https://github.com/PX4/Firmware/pull/7366

I and the tesflights team (when they flew the original fix) weren't initially able to reproduce it locally because it would only trigger depending on sensor startup timing (happens around 1/10 times). Sorry about all the drama regarding this fix :P

We still need QGC to force a reboot on setting mag and/or board orientation after calibration. @DonLakeFlyer @LorenzMeier

dagar, 2017-06-06:
Not sure if it's a bad idea, but you could have PX4 auto restart at these times with appropriate user notification.

closed 2017-06-07 by LorenzMeier