PX4/PX4-Autopilot #7362: Snapdragon - error finding params

ChristophTobler, 2017-06-06:
after [this commit](https://github.com/PX4/Firmware/commit/db816982cde07a19975b02bd2e7d63242d5d8f04), there is an error on the Snapdragon (see [output](http://pasted.co/632b6c12))

ChristophTobler, 2017-06-06:
@dagar I guess it has to do with this modification/restructuring https://github.com/PX4/Firmware/commit/24b26e53b92e655be8fabd7e07ac1c492b8ed0ec#diff-eff30623731b3af4359cadd642eb2645R208

dagar, 2017-06-06:
Fixed here - https://github.com/PX4/Firmware/pull/7359
I diffed all param output before and after across px4fmu-v3, snapdragon, etc and the only difference is mTecs sitting within fw_pos_ctrl unused is no longer included.

ChristophTobler, 2017-06-06:
@dagar Thanks! 
https://github.com/PX4/Firmware/pull/7359 fixes this issue but introduces another one https://github.com/PX4/Firmware/issues/7364

dagar, 2017-06-06:
I can revert the original commit if I can't figure out a fix shortly. I wasn't initially aware of parameter complexity on these platforms.

jywilson, 2017-06-06:
@dagar @ChristophTobler 

I also commented here:
https://github.com/PX4/Firmware/issues/7364#issuecomment-306536406

I read the comments over carefully and this sounds a lot like an issue we had before.  I describe the issue below, and how to check if it is happening again:

The aDSP build for Snapdragon uses a different toolchain than the AppsProc build.  The parameter files generated by each toolchain must be the same, in order for the parameter indexes to match when parameter uORB (see muORB module) messages are exchanged between the AppsProc and aDSP.  This means that if a new parameter is created in code that is compiled for the AppsProc, the parameter must also be included in the parameters file generated by the aDSP toolchain, even if the associated code is not compiled into the aDSP .so files.

You can check if this is a problem with the code changes in this PR by comparing the contents of parameters.xml file in the AppsProc and aDSP build directories generated under the Firmware directory.  The two files must be identical.  The build directories will be named "build_posix_{...}" and "build_qurt_{...}".  The build_posix tree contains AppsProc binaries and the build_qurt tree contains aDSP binaries.

Jim W.

dagar, 2017-06-12:
Fixed by https://github.com/PX4/Firmware/pull/7359

closed 2017-06-12 by dagar